pageauc/speed-camera #92: sudden change in sqpx while taking car speed

peterb99, 2021-06-25:
Im puzzled by a phenomenon I regularly see, significant change in the sqpx size while tracking a car, seems to happen more frequent by large objects,
 speed_camera Reset- event_timer 6.63>0.40 sec Exceeded
2021-06-25 17:11:41 INFO     speed_camera New  - 0/6 xy(29,25) Start New Track
2021-06-25 17:11:41 INFO     speed_camera  Add - 1/6 xy(38,24) 24.33 mph D=9/50 C=3 75x36=1409 sqpx L2R
2021-06-25 17:11:41 INFO     speed_camera  Add - 2/6 xy(47,24) 25.97 mph D=9/50 C=3 74x36=1386 sqpx L2R
2021-06-25 17:11:41 INFO     speed_camera  Add - 3/6 xy(55,24) 25.75 mph D=8/50 C=2 75x36=1371 sqpx L2R
2021-06-25 17:11:41 INFO     speed_camera  Add - 4/6 xy(64,26) 25.43 mph D=9/50 C=2 74x34=1307 sqpx L2R
2021-06-25 17:11:41 INFO     speed_camera  Add - 5/6 xy(100,37) 42.48 mph D=36/50 C=3 46x23=688 sqpx L2R
2021-06-25 17:11:41 INFO     speed_camera  Add - 6/6 xy(81,26) 44.64 mph D=19/50 C=2 73x34=1337 sqpx R2L

It some times triggers negative x pixel change between the last two intervals.

[image]
Can any settings be tweaked to circumvent this?

pageauc, 2021-06-25:
Yes that can happen due to a shift in largest contour.  I try to filter
some of those out in the code but it cannot eliminate the problem completely
due to the way opencv discovers contours.  My code does not do object
recognition due to speed of RPI.  This code was developed
by myself just for fun and is not intended to be perfect in all
situations.  The code works best if the vehicles do are not too close to
the camera where contours can easily move location on the moving object. Eg
from front to back or middle due to lighting or other changes that change
the location of the largest contour.

pageauc, 2021-06-25:
Note in the data you can see the second last contour was half the size of
the other contours eg 688 vs 1300+ sq pixels
This may have been due to lighting change or vehicle bodywork.  The C= is
the number of contours found.  It is possible
to join overlapping contours into one larger contour.   My Camera is
farther away from the road and I can get C=1 for the
track so better chance of accuracy although the contour can still shift of
the moving object.
Hope this explanation helps. Let me know if you have any ideas
Claude ...

closed 2021-06-26 by pageauc