pageauc/speed-camera #93: Libcamera Support

smyers75, 2021-06-27:
All of the new Higher Resolution Raspiberry Pi4 Cameras use Libcamera and do not support RaspiVid or RaspiStill
Is there any change of getting Speed Camera to work with Libcamera?

Example is the new ArduCam Pivariety IMX298 21MP and 16MP Cameras
Neither of these new cams currently work with Speed Camera

It would be an extra nice feature to be able to capture the license plate of the Speeders

pageauc, 2021-06-27:
Speed camera supports USB cameras,  IP Cameras through opencv and pi-camera
modules through pi-camera python library

Best performance is with a 320x240 image stream image size.  Using an
expensive camera module with limited software support seems like a lot of
work
for little gain when you can just use a relatively cheap rpi camera module
supported by python pi-camera software or and inexpensive
USB or IP camera.  I did look at the https://libcamera.org/index.html

https://www.raspberrypi.org/forums/viewtopic.php?t=273018&start=25

That said if the arducam is supported by opencv or can be configured as an
IP camera, then speed camera should work.

smyers75, 2021-06-27:
Sorry, I know "Github Issues" is not a discussion forum, but I'm unsure where else to say this.
Thank you for taking the time to look into libcamera as it seems pi is heading to it because it is open, unlike MMAL/picamera
Maybe the best solution for capturing details of excessively speeding vehicles would be two cameras, one at 320x240 to capture/log the speed of the object and another that is triggered by a setpoint to capture a HiRes photo of what just happened. (or maybe the HiRes comes 1st, then is overwritten if not past setpoint.)

closed 2021-06-27 by smyers75

pageauc, 2021-06-27:
It takes a few seconds for a camera to warm up.  If switching from a stream
to camera still mode maybe 1-2 seconds to ensure camera can get white
balance.

I have written a program called panopi  https://github.com/pageauc/panopi
that can simultaneously take multiple hi res images on multiple
pi-camera RPI's. I have used up to five.. A message is sent to each pano
camera to schedule the next timelapse event.  Once taken, images are  sent
back to a hub via zmq transmission protocol.  Hub can be one of the camera
RPI's or a separate computer.  The Hub then stitches the images into a
panorama image.  Taking images simultaneously means moving objects don't
cause a problem.  Triggering could also be by remote gpio, physical gpio or
some other network messaging.  I liked zmq.
Anyway
A similar setup can control one or more computers to trigger a hires image
and send it back if required.  You would need to trigger on the first
motion track event to allow time for camera warmup.  There are also RPI
camera modules that allow connecting multiple pi cameras.  You could also
just connect multiple USB camera or trigger a regular camera via serial
connection if supported.

I just program as one of my retirement pastimes for fun..  Taught myself
python but did have a background in C,, Pascal,, Basic, bash,, Etc.  I like
python but sometimes you love to hate it.
Claude ...