pageauc/speed-camera #94: Advice on tuning for cars

bushibot, 2021-07-01:
I wonder if there might be a better place to ask? but I'm having some tuning configuration challenges. 
I need to catch vehicles moving L-R, R-L, but I find the tracking really prefers people and cars across the street (so smaller objects?). This issue got worse when I moved picam720 60fps.
Despite the wiki and very detailed information I'm struggling with what to adjust to make it want the larger closer cars moving vs. everything else. I calibrated for vehicle in mm and pixel and set a speed rule 19mph, but my signal not noise ratio is still way to high
http://speed-car:8080/images/ car across 
http://speed-car:8080/images/ person L-R
http://speed-car:8080/images/ leafs (ARGH)

I'm 100% sure this is just me not grasping what effects what in the functions.

pageauc, 2021-07-01:
Upgrade speed camera to the latest version using menubox.sh UPGRADE menu
pick

You can use the config.py setting to set one or both of the variables
below.  These can be changed to tune the system.

MIN_AREA = 200

and or

max_speed_over = 10
Monitor log activity to see what pedestrians average square pixels value
and speed are.
Set MIN_SREA = value and or max_speed_over variables close to log values.

You can also view historical data by enabling csv logging.

in config.py set variable below

log_data_to_CSV = True

Runs speed camera for a while to collect data and update the speed-cam.csv

Then run the ./makehtml.py script. This can
also be run from menuubox.sh by running the HTML menu pick and viewing
results in the web browser interface under html directory
Note this report defaults to last 100 records but can be increased by
editing the variable HTML_MAX_FILES = 100 in the makehtml.py script
Note I am looking at putting makehtml.py settings in config.py

IMPORTANT: Try not to make the MIN_AREA variable too large since you may
eliminate some vehicles that are tracked with smaller contours due to
lighting or style.
Eg sometimes a wheel, trunk or other smaller areas can be tracked.

Give that a try

Also I have found I get best results by using a smaller stream image size
that reduces the amount of opencv processing.  Default is 320x240 for
webcam and pi-camera

Claude ...

pageauc, 2021-07-02:
Added information to wiki
https://github.com/pageauc/speed-camera/wiki/How-to-Tune-Motion-Tracking-Settings#tune-out-smaller-moving-objects

closed 2021-07-02 by pageauc

bushibot, 2021-07-02:
> Upgrade speed camera to the latest version using menubox.sh UPGRADE menu pick You can use the config.py setting to set one or both of the variables below. These can be changed to tune the system. [...]
Thank you.
I messed my setting up enough that I stated over. I could watch a car go through the tracking area and nothing would get locked 😤😂. Yes, oddly lower res seemed to work better for me too. Not as good for identifying the car though.
Upgraded.
Could I trouble you expand slightly on how the functions behave by example.
If I shrink or grow the MIN_AREA, how dose that effect the tracking? Helping or hurting the contour it is looking for? I tried making a much larger car sized block but that failed to get cars but would still get bikes across the street.
Speed is pretty clear, but a lot of the other variables are not, at least to me, as far as cause and effect.

Amazing when it it’s working.

pageauc, 2021-07-02:
At approx line 1180 in speed-cam.py per code below is the only real
programming spot (others are just for info) that MIN_AREA is used..  Moving
objects can]
be made of more than one contour due lighting, reflection, shape, Etc  You
can see the number of contours in the tracking logs when running
speed-cam.py from a console with verbose turned on. In the log you will see
C= that shows how many contours were found.  Ideally this should be one. If
the moving object is too close it can generate multiple contours.  At any
rate the program cycles through the contours and checks initially if it is
larger than the MIN_AREA otherwise the contour is ignored.  It then looks
for the largest moving contour and that is the one it tracks.  Determines
the center x,y position of the contour center.  That determines a single
point..and a track is made up of a series of points that get filtered to
make sure they are reasonably close to the previous point otherwise it is
considered OUT of range and rejected.  This can happen for lots of reasons
like more than one object moving or the location of the largest contour
shifting position on the moving object.  This is what can cause erratic
speed shifts especially if the camera is too close so that the moving
object takes up a large part of the camera view.  You will get this with a
camera too close or things like trucks or like myself cutting the grass or
birds flying close to camera and inside the tracking window.   At any rate
motion tracking follows the moving object where each point has to be within
a certain number of pixels from the previous position.  Contour movements
on the object can be smoothed out over a longer track.

So MIN_AREA rejects all contours less than the setting.  Most of the
problems users have is the camera is less than approx 15m or 50f from the
calibration zone or they try to set the camera stream resolution too high.
Hope this helps.

>         if contours
>             total_contours = len(contours)
>             motion_found = False
>             biggest_area = MIN_AREA
>             for c in contours:
>                 # get area of contour
>                 found_area = cv2.contourArea(c)
>                 if found_area > biggest_area:
>                     (x, y, w, h) = cv2.boundingRect(c)
>                     # check if object contour is completely within crop
> area
>                     if (x > x_buf and x + w < x_right - x_left - x_buf):
>
>                         track_x = x
>                         track_y = y
>                         track_w = w  # movement width of object contour
>                         track_h = h  # movement height of object contour
>                         motion_found = True
>                         biggest_area = found_area
>
>>                      cur_track_time = time.time() # record cur track time
>
>