pageauc/speed-camera #95: Plugin webcam720 cannot calibrate

jameshfischer, 2021-07-17:
using default low-rez 240p plugin, all is well, but changing to webcam720, and appropriately changing parameters within that file results in calibration mode that never see any motion at all.

Motion is certainly detected, but it does not result in a trigger.

What would be some "good guess" settings for detection box size with the webcam720 plugin?

pageauc, 2021-07-17:
Run ./speed-cam.py in ssh or terminal session and view the logs.  You
should see what is causing the issue.
I suspect you are getting a lot of track event timeouts.  The default is
0.3 sec.  Opencv just has to process too much data
in a 720 image.  You could try reducing the tracking window size by
uncommenting the following config.py settings
This will override the auto calculation of motion tracking window.

# x_left = 50          # uncomment and change values to override auto
> calculate
> # x_right = 300        # uncomment and change values to override auto
> calculate
>
> # Allow user to customize the motion crop area (height) y values
> # If variables not found then values will be set automatically base on
> image size.
> # y_upper = 60         # uncomment and change values to override auto
> calculate
> y_lower = 150          # uncomment and change values to override auto
> calculate
>

then make sure following setting is set to True (default)

image_show_motion_area = True
>

720 resolution image is 1920 x 720 so start with a very small tracking
window size. roughly equivalent to what you
would get with the autocalculated 320x240 resolution.  Numbers have been
changed to position window near
the center of the camera image.  You can change these to suit.

x_left = 600

x_right = 850

y_upper = 300

y_lower = 390

You will need to repoint the camera to put the motion area in  the tracking
window.  I sometimes move my hand in front to trigger motion tracking
so I can see where the window is pointed
Depending on what logs say you can tune other parameters per comments These
will be in the Motion Tracking Settings section of
the config.py file
eg
track_counter = 4

Make sure the camera is at least approx 50 ft away or more or you will have
problems tracking since a objects will take up so much of the camera view.

Give these suggestions a try and let me know how you make out

Also see
https://github.com/pageauc/speed-camera/wiki/How-to-Tune-Motion-Tracking-Settings

Regards Claude ...

>

jameshfischer, 2021-07-18:
I do not think that a Pi-3B can handle 720p, and the graphics processing required even with quad-core processors. 
The pi crashes and burns, as shown below:

First, the initial debug messages when in calibration mode

[image]

Then, the capture and resulting crash
[image]

pageauc, 2021-07-18:
I am testing on a RPI 3B overclocked to 900 using raspi-config menu

I had trouble with 720 but am testing at 1024x704 resolution using RPI
camera.  Not sure if this would work on a web cam though
Also run under python3 but works OK on python2 as well.  Just seeing if
there was a difference. Working with higher
resolutions is definitely finicky and does have to be tuned to suit.
Opencv processing seems to be the bottleneck

Save your old config.py eg cp config.py config.py.mine the copy the
attached to config.py

You should not normally get a kernel panic error.  Might be a problem
with SD card but could also be the USB Webcam driver
Try a full
sudo apt update -y
sudo apt upgrade -y

Not sure if you are using the latest Buster or an earlier Build.

Anyway the attached config.py settings are a work in progress for my own
curiosity.  I normally run mostly stock
config.py settings and this runs on a OK on a single core and quadcore RPI.
with mild overclock at 800

Keep me up to date.  You can always go back to the default settings that
work pretty well even on low level RPI's

jameshfischer, 2021-07-18:
I went back to the 240p configuration, as low-quality photos are not a deal-breaker.  (The goal here is to document speeds of vehicles passing grandmother's house (102 yrs) as some cars roar up the street at 50mph+ in a 25mph town, and a speed bump is needed to protect the little old ladies who live on the street. )

But I am getting inconsistent results, and multiple speed estimates for the same car... some wildly variant.

The variations in speed readings are concerning, but when I drove my own car past the camera, it seemed within 1mph of my speedometer.

For example, 38mph, 30mph, and 30mph for the same car within less than a second:
[image]
[image]
[image]

and 19mph, then 143mph:
[image]
[image]

25mph, then 17mph
[image]
[image]

Below is the startup, config, and a typical capture
pi@speedcam:~/speed-camera $ ./speed-cam.py
----------------------------------------------------------------------
speed-cam.py 11.09  written by Claude Pageau
Motion Track Largest Moving Object and Calculate Speed per Calibration.
----------------------------------------------------------------------
Loading  Wait ...
2021-07-18 19:22:24 INFO     <module>   pluginEnabled - loading pluginName /home/pi/speed-camera/plugins/webcam240.py
2021-07-18 19:22:24 INFO     <module>   Copy /home/pi/speed-camera/plugins/webcam240.py to /home/pi/speed-camera/plugins/current.py
2021-07-18 19:22:24 INFO     <module>   Import Plugin /home/pi/speed-camera/plugins/webcam240.py
2021-07-18 19:22:28 INFO     <module>   Initializing USB Web Camera Try .. 1
----------------------------------------------------------------------
Note: To Send Full Output to File Use command
python -u ./speed-cam.py | tee -a log.txt
Set log_data_to_file=True to Send speed_Data to CSV File speed-cam.log
----------------------------------------------------------------------

Debug Messages .. verbose=True  display_fps=False calibrate=False
                  show_out_range=True
Plugins ......... pluginEnable=True  pluginName=webcam240
Calibration ..... cal_obj_px_L2R=105 px  cal_obj_mm_L2R=4503 mm  speed_conv_L2R=0.09395
                  cal_obj_px_R2L=105 px  cal_obj_mm_R2L=4503 mm  speed_conv_R2L=0.09395
                  (Change Settings in /home/pi/speed-camera/plugins/webcam240.py)
Logging ......... Log_data_to_CSV=True  log_filename=speed-cam.csv (CSV format)
                  loggingToFile=False  logFilePath=speed-cam.log
                  SQLITE3 DB_PATH=/home/pi/speed-camera/data/speed_cam.db  DB_TABLE=speed
Speed Trigger ... Log only if max_speed_over > 10 mph
                  and track_counter >= 8 consecutive motion events
Exclude Events .. If  x_diff_min < 1 or x_diff_max > 30 px
                  If  y_upper < 100 or y_lower > 150 px
                  or  x_left < 90 or x_right > 300 px
                  If  max_speed_over < 10 mph
                  If  event_timeout > 0.50 seconds Start New Track
                  track_timeout=0.20 sec wait after Track Ends (avoid retrack of same object)
Speed Photo ..... Size=960x720 px  image_bigger=3.0  rotation=0  VFlip=True  HFlip=True
                  image_path=media/images  image_Prefix=speed-
                  image_font_size=12 px high  image_text_bottom=True
                  image_jpeg_quality=98  image_jpeg_optimize=True
Motion Settings . Size=320x240 px  px_to_kph_L2R=0.151200  px_to_kph_R2L=0.151200 speed_units=mph
                  CAM_LOCATION= None
OpenCV Settings . MIN_AREA=100 sq-px  BLUR_SIZE=10  THRESHOLD_SENSITIVITY=20  CIRCLE_SIZE=5 px
                  WINDOW_BIGGER=1 gui_window_on=False (Display OpenCV Status Windows on GUI Desktop)
                  CAMERA_FRAMERATE=30 fps video stream speed
Sub-Directories . imageSubDirMaxHours=0 (0=off)  imageSubDirMaxFiles=2000 (0=off)
                  imageRecentDir=media/recent imageRecentMax=100 (0=off)
Disk Space  ..... Disabled - spaceTimerHrs=0  Manage Target Free Disk Space. Delete Oldest jpg Files
                  spaceTimerHrs=0 (0=Off) Target spaceFreeMB=500 (min=100 MB)

----------------------------------------------------------------------
2021-07-18 19:22:32 INFO     isSQLite3  Success: File is sqlite3 Format /home/pi/speed-camera/data/speed_cam.db
2021-07-18 19:22:32 INFO     db_check   Success: sqlite3 Connected to DB /home/pi/speed-camera/data/speed_cam.db
2021-07-18 19:22:32 INFO     speed_camera sqlite3 DB is Open /home/pi/speed-camera/data/speed_cam.db
2021-07-18 19:22:32 INFO     speed_notify Plugin Enabled per pluginName=webcam240
2021-07-18 19:22:32 INFO     speed_notify Logging to Console per Variable verbose=True
2021-07-18 19:22:32 INFO     speed_notify Press ctrl-c in this terminal session to Quit
2021-07-18 19:22:32 INFO     speed_notify Begin Motion Tracking .....
2021-07-18 19:23:05 INFO     speed_camera Reset- event_timer 32.44>0.50 sec Exceeded
----------------------------------------------------------------------
2021-07-18 19:23:05 INFO     speed_camera New  - 0/8 xy(146,41) Start New Track
2021-07-18 19:23:05 INFO     speed_camera  Add - 1/8 xy(134,41) 35.61 mph D=12/30 C=2 27x9=165 sqpx R2L
2021-07-18 19:23:05 INFO     speed_camera  Add - 2/8 xy(122,38) 33.58 mph D=12/30 C=2 28x12=190 sqpx R2L
2021-07-18 19:23:05 INFO     speed_camera  Add - 3/8 xy(110,36) 33.90 mph D=12/30 C=3 29x14=246 sqpx R2L
2021-07-18 19:23:05 INFO     speed_camera  Add - 4/8 xy(98,25) 34.28 mph D=12/30 C=2 85x25=1243 sqpx R2L
2021-07-18 19:23:05 INFO     speed_camera  Add - 5/8 xy(87,26) 33.79 mph D=11/30 C=2 83x24=1216 sqpx R2L
[... cut ...]
2021-07-18 19:23:05 INFO     log_to_csv    CSV - Appended Data into /home/pi/speed-camera/speed-cam.csv
2021-07-18 19:23:05 INFO     saveRecent    symlink /home/pi/speed-camera/media/recent/speed-20210718-1923056.jpg
2021-07-18 19:23:05 INFO     speed_camera End  - R2L Ave Speed 31.6 mph Tracked 89 px in 0.267 sec Calib 105px 4503mm
----------------------------------------------------------------------
2021-07-18 19:23:05 INFO     speed_camera track_timeout 0.20 sec Sleep to Avoid Tracking Same Object Multiple Times.

pageauc, 2021-07-19:
> 2021-07-18 19:22:24 INFO pluginEnabled - loading pluginName
> 2021-07-18 19:22:24 INFO Copy /home/pi/speed-camera/plugins/webcam240.py
> 2021-07-18 19:22:24 INFO Import Plugin
> 2021-07-18 19:22:28 INFO Initializing USB Web Camera Try .. 1 Note: To

speed camera tracks the largest contour of moving objects in the camera
tracking window.  If the camera is too close and the moving object is
large, then
the largest contour can move around to different locations on the moving
object. C= is the number of contours. This will cause speed variations.
Very large objects like trucks or trailers or something very close to
the camera like a bird can be a challenge as well.

A number of settings for OUT of Range conditions can be used but can only
go so far.  Longer track counts can increase accuracy.  I have also found
that colour and lighting can affect opencv tracking.  Eg a grey object on a
grey background can be more difficult to track than a white object on a
dark background.  Sometimes lighting, object shape, reflections can be
factors.

There are also factors like varying distances within the camera view.  Eg
the distance to the far right and left is farther away than in the middle
of the camera view.

I wrote Speed camera as a fun retirement programming project.  It is
basically an object tracker and the speed accuracy can vary.  Best results
are when the moving objects are smaller in the camera view.  Eg 75 to 100
ft from camera.
Sometimes the problem is unrealistic expectations.  Even radar systems can
have issues.

Hope this explanation helps.

jameshfischer, 2021-07-19:
So, increase the "track count" parameter, and tweak that setting to avoid double-counting the same car?

bushibot, 2021-07-27:
'. Eg 75 to 100 ft from camera. Sometimes the problem is unrealistic expectations. Even radar systems can have issues. Hope this explanation helps.'

This has been challenge for me, my camera sits about 35ft from the middle of the adjacent road. Does angle affect the reading? I'm wondering if I put it up on the the 3rd floor window angled it down at the street to increase distance if it  would help. Similar projects seem to require a flat level view to the street though.

pageauc, 2021-07-27:
pointing camera down from an upper floor should work to increase the
distance. It needs to be parallel with the road otherwise angle will mean
varying distance in camera vew. and can give inconsistent results.