pageauc/speed-camera #96: trying to setup crontab to start camera at 6am and stop at 2000 but doesn't work.

johncblacker, 2021-07-18:
I'm trying to setup crontab to start camera at 0600, stop at 2000 but it's not working - never continues running.
My crontab entry:
0 6 * * * /home/pi/git-projects/speed-camera/speed-cam.sh start >> /home/pi/git-projects/speed-camera/speedcam.log
59 20 * * * /home/pi/git-projects/speed-camera/speed-cam.sh stop >> /home/pi/git-projects/speed-camera/speedcam.log
My speedcam.log shows:
START   - Start speed-cam.py in Background ...
STATUS  - speed-cam.py is Running ...
STATUS  - PID is 31430
INFO    - To Stop speed-cam.py execute command below
INFO    - /home/pi/git-projects/speed-camera/speed-cam.sh stop
Done
/home/pi/git-projects/speed-camera/speed-cam.sh ver 6.00 written by Claude Pageau
But if I check for running speed-cam after the start, it isn't there.  What might I be doing wrong?
I also tried starting it after reboot with:
@reboot cd /home/pi/git-projects/speed-camera && ./speed-cam.sh start >/home/pi/git-projects/speed-camera/cronlog.log but it doesn't start, not running when I check.  Maybe some guidance in trying to find out what's wrong?

pageauc, 2021-07-18:
Edit speed-cam.sh script and change the *progpath *variable to path where
speed-cam.py is located.  The shell script is hard coded to
/home/pi/speed-camera so change to

*progpath="/home/pi/git-projects/speed-camera/"*

This is the top of

> #!/bin/bash
> # This script will run the speed-cam.py as a background task
> # You will then be able close the terminal session.  To auto start
> # Add the following command to the /etc/rc.local
> # /home/pi/speed-camera/speed-cam.sh start
>
> progpath="/home/pi/speed-camera"
> progname="speed-cam.py"
>

make sure speed-cam.sh works from command line

Let me know if this resolves your problem.

johncblacker, 2021-07-19:
I had already made the directory that you suggest early on in my testing.  I will try changing rc.local as you suggest to take 

care of the reboot problem; but that still leaves the issue regarding starting the application and stopping it via cron.  I’ve

researched that online and found nothing that suggests I’m doing anything wrong, so it’s still a mystery.  I even changed

from using the shell script to simply trying to start the speed-cam.py program itself in the crontab and looking at my output

from that and, there again, I see where it was supposedly started, yet when I check via ./speed-cam.sh status it indicates it

is not running.  I also used “ps -ef | grep <pid> “ and “ps -ef | grep speed” to check the status of the app and it all cases 

it is not running.  So, for whatever reason, it might have gotten started, then ended?  I’m searching for any ideas as to how

to pursue finding the error.  I have used simply dummy test.sh scripts in the crontab and they execute fine.

 

So, thanks for your suggestion.  I will continue to pursue this, but if you have any further ideas regarding the crontab issue

I welcome them and will try anything.

 

Thanks again.

 

<http://www.youtube.com/user/pageaucp>

johncblacker, 2021-07-19:
So, put the startup command in rc.local, rebooted and the output from the startup said it failed to import cv2!  So, at whatever level it was at when

it ran, it couldn’t perform the import; however, I don’t have any problems running after startup is complete.  Wonder if there’s a way to only execute

the command at a specific “run level” – I’ll have to research that.

 

<http://www.youtube.com/user/pageaucp>

johncblacker, 2021-07-19:
********** update 14:54 est ******************

Success running as a service.  I wasn’t able to make it work by starting via the speed-cam.sh script so I reverted to just

using ‘python speed-cam.py’ after first changing the settings so that the cv windows weren’t being displayed…that doesn’t work

when running as a service; but, who cares? If you want to see you can go to the webserver and display the current image(s) which 

is actually better because they have the speed displayed on them.

 

So, I now have something that’ll work a reboot time, all I have to do is issued ‘sudo systemctl enable speedcam.service’

Here is a link to the page I fashioned my speedcam.service after:

                systemd - Raspberry Pi Documentation <https://www.raspberrypi.org/documentation/linux/usage/systemd.md> 

Here’s my “unit file” speedcam.service contents:

 

[Unit]

Description=Speed camera

After=network.target

 

[Service]

ExecStart=/bin/bash -c '\

python /home/pi/git-projects/speed-camera/speed-cam.py > /home/pi/git-projects/speed-camera/script.txt 2>&1'

WorkingDirectory=/home/pi/git-projects/speed-camera

StandardOutput=inherit

StandardError=inherit

Restart=on-failure

RestartSec=30s

User=pi

 

[Install]

WantedBy=multi-user.target

 

That file is in /etc/systemd/system named speedcam.service

 

Now, I’m going to tackle the up/down at a specific time of day…

 

cheers!

 

 

********** Update ****************

switched over from crontab to starting this as a service.  That would then start immediately after reboot; and by having it a service, I can simply put

systemctl restart speedcam.service in crontab – which I’m sure will work.

 

But (there’s always a but, isn’t there) when I tried to test start the service, I get some messages in the log which indicate a problem:

 

Started Speed camera.

Jul 19 13:57:36 client1 systemd[1]: speedcam.service: Main process exited, code=exited, status=1/FAILURE

Jul 19 13:57:36 client1 systemd[1]: speedcam.service: Failed with result 'exit-code'.

 

So, there appears to be a problem.

 

<http://www.youtube.com/user/pageaucp>

closed 2021-07-19 by johncblacker

johncblacker, 2021-07-19:
After converting to a service, the process of starting at boot is simplified by simply "systemctl enable speedcam.service" and commands in crontab simply become: "systemctl start speedcam.service" and "systemctl stop speedcam.service" and I verified those work.  Cron is a fine service, but difficult to debug when something doesn't work; although by having your scripts or command write output to a <script>.log file it's possible to figure it out.  After seeing what can be done by making a "job" a service, I'm sold on it as a wonderful solution and in addition, it makes your crontab commands simpler.  By puttiing something in rc.local, you face the problem that because the boot process happens in stages (or levels) your command can end up running at a level that doesn't have the necessary resources to be executed and it fails.  In the case of the service, the "WantedBy" parameter takes care of that - there are a number of "targets" and they can be listed with the systemctl cmd (systemctl list-units --type target).

pageauc, 2021-07-20:
You might want to try  command below in your crontab

su pi -c "/home/pi/git-projects/speed-camera/speed-cam.sh start"

Let me know if this works

johncblacker, 2021-07-20:
No, didn’t work, here’s output from syslog:

Jul 20 14:15:01 client1 cron[548]: (pi) RELOAD (crontabs/pi)

Jul 20 14:16:01 client1 CRON[12627]: (pi) CMD (su pi -c"/home/pit/git-projects/speed-camera/speed-cam.sh start")

Jul 20 14:16:02 client1 postfix/pickup[26412]: 980902A0EFE: uid=1000 from=<pi>

Jul 20 14:16:02 client1 postfix/cleanup[12631]: 980902A0EFE: ***@***.***>

Jul 20 14:16:02 client1 postfix/qmgr[1016]: 980902A0EFE: ***@***.***>, size=629, nrcpt=1 (queue active)

Jul 20 14:16:02 client1 postfix/local[12633]: 980902A0EFE: ***@***.***>, orig_to=<pi>, relay=local, delay=0.26, delays=0.17/0.06/0/0.04, dsn=2.0.0, status=sent (delivered to mailbox)

************************************

 

<http://www.youtube.com/user/pageaucp>

johncblacker, 2021-07-20:
It did work from the terminal, though…but it did ask for my password so maybe that’s the issue with it not working?

 

<http://www.youtube.com/user/pageaucp>

johncblacker, 2021-07-20:
Hang on…trying again, I made a typo in the command.  Retrying…

 

<http://www.youtube.com/user/pageaucp>

johncblacker, 2021-07-20:
Tried again without the typo and still didn’t work.  Going to try again like this:  sudo -s “/home/pi/git-projects/speed-camera/speed-cam.sh start”

and see what happens.

 

<http://www.youtube.com/user/pageaucp>

johncblacker, 2021-07-20:
Tried again, no luck that time either…

 

<http://www.youtube.com/user/pageaucp>