pageauc/speed-camera #97: Variable in config.py to specify media root directory

peterb99, 2021-08-12:
I would like to keep the media directory and it contents on a separate mounted file system to prevent the system from lockup when the media images fill up the root file system and to reduce wear and tear off the root Sd card in the pi.

I've tried to mount a separate file system on $HOME/speed-cam/media, but ran in file and directory access permissions issues with that. Look like a mounted file system at the root of the main system is needed.

Can a variable be added to specify the media full directory path like /speed/media , where /speed is the mount point of the extra file system?

pageauc, 2021-08-12:
I will write a wiki on doing this.  Currently I use small SSD drives for
external storage  eg
https://www.amazon.ca/gp/product/B07GXHPR6R/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

I move all of the ~/speed-camera folder to the SSD and create a soft link
to the mount location.  This allows moving the drive to another RPI and not
loosing any data or configuration if the rpi system sd card fails,

*Step 1 *
Install and mount external storage device per
https://www.raspberrypi.org/documentation/computers/configuration.html#external-storage-configuration
I usually mount to /home/pi/ssd-mnt but you can mount to a folder of your
choice.  For this example I will assume mount point is ssd-mnt

sudo apt update
> sudo apt upgrade
> sudo apt install ntfs-3g   # this should already be installed

plug in usb drive

Here is an example of one of my ssd drives plugged in

sudo blkid
>

/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="6284-658D"
TYPE="vfat" PARTUUID="254cd2b1-01
/dev/mmcblk0p2: LABEL="rootfs" UUID="3a324232-335f-4617-84c3-d4889840dc93"
TYPE="ext4" PARTUUID="254cd2b1-02"
/dev/mmcblk0: PTUUID="254cd2b1" PTTYPE="dos"
/dev/sda1: UUID="7476C11B76C0DECC" TYPE="ntfs" PARTUUID="cc7444a5-01"

My ssd drive is located at /dev/sda1 per above.
Below is an example of my /etc/fstab entry eg

sudo nano /etc/fstab

 ctrl-x y to save changes.  Note you will have to use disk uuid for your
drive or just use /dev/sda1 but uuid is safer

 UUID=7476C11B76C0DECC /home/pi/ssd-mnt ntfs-3g
> defaults,user,uid=1000,gid=1000,nofail,noatime 0 0

You can reboot or just mount manually per

sudo mount ssd-mnt

Make sure you can access the ssd drive and everything works OK including
permissions.  If you have problems
review the rpi org doc

*Step 2*
Stop speed camera and webserver using menubox.sh or the shell scripts
Move speed-camera folder to external storage device and create symbolic
link per steps below
Note this could take some time depending on size of media folder

cd ~
> mv speed-camera /home/pi/ssd-mnt/
> ls -l ssd-mnt
> ln -s /home/pi/ssd-mnt/speed-camera speed-camera

ls -l speed-camera

speed camera link should be owned by pi per eg below

lrwxrwxrwx 1 pi pi 29 Aug 12 09:37 speed-camera ->
/home/pi/ssd-mnt/speed-camera

double check permissions in speed-camera

cd ~/speed-camera
ls -l

You should see all files and folders owned by pi user.  If not check your
fstab mount

*Step 3*
Restart speed camera using menubox.sh or the specific shell scripts.

Let me know your progress
This should also work on any ntfs storage device.  In some cases I reformat
drive as ext4 using gparted utilit

I will make this a wiki.  Please provide feedback in case I have missed
something.

Claude

https://www.raspberrypi.org/documentation/computers/configuration.html#external-storage-configuration

pageauc, 2021-08-12:
I have created a wiki here
https://github.com/pageauc/speed-camera/wiki/How-to-move-speed-camera-to-an-external-storage-device

Note change the fstab similar to below.  I was having execute permission
issuel.

UUID=7476C11B76C0DECC /home/pi/ssd-mnt ntfs-3g
auto,uid=1000,exec,rw,async,noatime,nofail       0 0

peterb99, 2021-08-12:
I ran into the exec problem too, when using ext4 as filesystem type uid= and gid= are not supported, I also added x-systemd.device-timeout=1 to prevent a system hangup if the mount fails during boot.

pageauc, 2021-08-12:
I have not previously run across the exec problem so am thinking is a
recent security update or something.

Can you send me a copy of your fstab.  I am looking at adding the systemd
device-timeout=
This is what is in the wiki

UUID=7476C11B76C0DECC /home/pi/ssd-mnt ntfs-3g
> auto,uid=1000,exec,rw,async,noatime,nofail,x-systemd.device-timeout=1ms  0 0

peterb99, 2021-08-13:
Entry I'm using
UUID=08abe039-e7b7-4af5-ac09-37e67d994556 /home/pi/speed-ssd ext4 defaults,user,exec,nofail,x-systemd.device-timeout=1,noatime 0 0

closed 2021-08-23 by pageauc