pageauc/speed-camera #100: How to mask part of the picture

amcaw, 2021-09-30:
Thanks for this project, great stuff, works like a charm ! It is more a question than an issue : is there a way to mask part of the picture so I only capture the motion crop area (i.e.only the part inside of the "red box") ?

So : v1.jpg : what I get. v2.jpg : what I want
[image]
[image]

Should I add something in config.py in Pi Camera Settings ?

Thanks !

pageauc, 2021-09-30:
Yes you can override the automatic speed motion window in config.py using the settings below.  Note uncomment x_left, x_right and y_upper. variables.  y_lower is already uncommented.  Set these variables to the location that you want.  0,0 is the top left corner of the full size image.  Make sure to keep settings within the full image dimensions You can also move camera position as well..  This should do what you want.  Let me know if this answers you question.  

See config.py settings and comments below
```
# Note: To see motion tracking crop area on images, Set variable image_show_motion_area = True

# Allow user to customize the motion crop area (width) x values
# If variable not found then values will be set automatically based on image size.
# x_left = 50          # uncomment and change values to override auto calculate
# x_right = 300        # uncomment and change values to override auto calculate

# Allow user to customize the motion crop area (height) y values
# If variables not found then values will be set automatically base on image size.
# y_upper = 60         # uncomment and change values to override auto calculate
y_lower = 150          # uncomment and change values to override auto calculate

```

amcaw, 2021-09-30:
Thanks a lot for your reply ! Your explanations and code seem to be related to the size of the crop area (the "red box" then). My question was about cropping the whole picture to keep only what's inside the red box (to hide houses in front of the camera and only "see" the road and the cars passing by). A bit like in the pics below.

I saw this in your code, but couldn't find how to make it work. But maybe it's not related to what I'm trying to do...

`# crop image to motion tracking area only
        try:
            image_crop = image[y_upper:y_lower, x_left:x_right]
            image_ok = True`

Thanks anyway, your tool is awesome. 

[image]
[image]

pageauc, 2021-09-30:
The crop area can be saved as an image although it would be smaller.  Note:
Only motion inside the box is tracked.  You can add a save of the cropped
image instead of the full image

eg at approx line 1301 change image2 to image_crop

                                # Resize and process previous image
>                                 # before saving to disk
>                                 *prev_image = image2*
>

change to

*prev_image = image_crop*

You can give that a try.  make sure config.py  variable   *image_text_on =
False *
since  image size will be smaller.

Let me know if this works

amcaw, 2021-10-01:
Yes, you are right, it does crop the picture as I wanted ! Only thing is the cropped area is focusing on the top left corner of the frame, so I guess I'll have to find a trick to make this area match the one of my "red box". Maybe by changing `image_crop = **image[y_upper:y_lower, x_left:x_right]**` ? With integer ?

amcaw, 2021-10-01:
Thanks, it's working with the explanations you provided !

closed 2021-10-01 by pageauc