pageauc/speed-camera #104: Program stops

PE1DDA, 2021-12-09:
Good day,

Thank you for this wonderful piece of SW.
I am using W10 - Python 3.10 (full install) running a rtsp feed from outdoor IP cam.
Everything seems to work fine but after a few "motion detections" it quits the program by below verbose output.
I hope you can help me with this for which many thanks in advance!

2021-12-09 12:33:04 INFO     speed_camera New  - 0/6 xy(357,36) Start New Track
2021-12-09 12:33:04 INFO     speed_camera  Add - 1/6 xy(348,35) 17.81 kph D=9/200 C=1 67x69=2919 sqpx R2L
2021-12-09 12:33:04 INFO     speed_camera  Add - 2/6 xy(337,32) 18.13 kph D=11/200 C=6 67x73=3119 sqpx R2L
2021-12-09 12:33:04 INFO     speed_camera  Add - 3/6 xy(323,30) 19.84 kph D=14/200 C=2 69x70=3065 sqpx R2L
2021-12-09 12:33:05 INFO     speed_camera  Add - 4/6 xy(314,33) 18.69 kph D=9/200 C=1 64x67=2782 sqpx R2L
2021-12-09 12:33:05 INFO     speed_camera  Add - 5/6 xy(306,31) 17.36 kph D=8/200 C=1 63x67=2790 sqpx R2L
2021-12-09 12:33:05 INFO     speed_camera  Add - 6/6 xy(301,29) 16.12 kph D=5/200 C=2 59x68=2561 sqpx R2L
2021-12-09 12:33:05 INFO     speed_camera  Saved media/images\speed-20211209-1108/speed-16-20211209-1233052.jpg
Traceback (most recent call last):
  File "C:\speed-camera\speed-cam.py", line 1722, in <module>
    speed_camera() # run main speed camera processing loop
  File "C:\speed-camera\speed-cam.py", line 1395, in speed_camera
    cv2.imwrite(filename, big_image, [int(cv2.IMWRITE_JPEG_QUALITY), image_jpeg_quality,
cv2.error: OpenCV(4.5.4) :-1: error: (-5:Bad argument) in function 'imwrite'
> Overload resolution failed:
>  - Can't parse 'params'. Sequence item with index 3 has a wrong type
>  - Can't parse 'params'. Sequence item with index 3 has a wrong type

pageauc, 2021-12-09:
> cv2.error: OpenCV(4.5.4) 👎 error: (-5:Bad argument) in function 'imwrite'

So you are getting images saved sometimes but after a while it errors out.
The line syntax is OK as far as I can see so I suspect problem with IP
camera image due to possible transmission error or something.

Can you try saving as .png format
Edit config.py and change

image_format = ".png"

Retry and let me know the result

Claude ...

pageauc, 2021-12-09:
Also is w10 mean windows 10 or is this a raspberry pi

pageauc, 2021-12-09:
Checked opencv 4.5.4 syntax for imwrite per
https://docs.opencv.org/4.5.4/d8/d6a/group__imgcodecs__flags.html#ga292d81be8d76901bff7988d18d2b42ac
I suspect a problem with the IP image stream.  If the jpg imwrite fails I
have added an exception to just save unoptimized image

speed-cam.py has been updated to version 11.22 on github

If the png workaround works then this new version should resolve your jpg
save issue but needs testing.

Please let me know results of your testing

Thanks Claude ...

PE1DDA, 2021-12-09:
Hello Claude,

Thank you for your prompt reply!
By the way, my name is Cees (pronounced like "case" as in suit_case).
PE1DDA = my licensed HAM call-sign :-) 

I'm on a Windows 10 (x64) machine (AMD Ryzen).
I use the webcam720 plugin on the substream of my IP camera (ethernet)
The stream from the IP camera looks like it's okay (I monitor on another channel 1080p mainstream with OnVif)
Thank you very very much! changing the file format to .png did the trick, no more closing python !!
Never thought of this and I'm happy with .png format..........
Tonight I will certainly give your updated file a try and will let you know the results tomorrow.

Once again thank you very much!

73's (that's HAM talk for all the best)

Cees

pageauc, 2021-12-09:
Your welcome

btw  were you getting some successful jpg saves before imwrite error or was
error right away.
Your ver of opencv 4.5.4 is brand new but the optimized imwrite syntax
appears to be correct and works on other opencv versions.  If error is not
immediate then
there may be something in the IP stream that triggers the error otherwise
let me know and I can do more research with opencv 4.5.4

Claude ....

PE1DDA, 2021-12-10:
Good day Claude,

The error occurred immediately and was there all along since I started this project on my Raspberry Pi 2B + 2 weeks ago with an old USB camera, to just give it a try for fun.

The only image ever saved was the one by: "align_cam_on = True"

I changed to my Windows machine because it has far more CPU power and I'm more familiar with all the "looks" and tools I have there (also a full Debian install on second boot).
Software is not really my thing, in fact I never heard of open CV before this project, I'm a hardware gui that loves to build electronics, hence this speed camera will be combined/integrated in Domoticz system already running for 5 years now on a Raspberry Pi zero W
To make a long story short: your new version of speed-cam.py solves it!
Now on .jpg without any problem :-) !!!

Thank you so much for your devotion and prompt action, now I only need to do some calibration and can start "catching" the idiots that do illegal street races on the road I'm living at, this is a 60 KM road but they start to double that for a minimum!

All the best and stay healthy!

Cees

closed 2021-12-16 by pageauc