pageauc/speed-camera #106: not using substream?

mlapaglia, 2021-12-28:
I am using RTSP with a camera trying to get speed-camera to analyze the sub stream, which is much lower resolution for more CPU friendly processing.

Regardless of what stream I pass in the URL (main stream is 1440p, substream is 704x480), the same amount of CPU is used.  When I look at the images showing up on the web server they are all 1440p images, not the smaller sub stream images.

my URL looks like `rtsp://username:password@<ip address>/cam/realmonitor?channel=1&subtype=2`

pageauc, 2021-12-29:
Speed camera will auto detect the RTSP image resolution and process that.
It does not use the stream resolution in the config.py because it does not
have control of the RTSP camera and that is why you are getting the RTSP
resolution as sent from the IP camera

At approx line 1684 in speed-cam.py The stream is read.  and the shape of
the image is read for the width and height.  For a physical PiCamera or USB
Webcam this will be the size as specified in the config.py for the
appropriate camera.  But if it is an RTSP camera is will read the stream
image size that has nothing to do with the config.py settings since the
camera is not physically attached

            test_img = vs.read()
            img_height, img_width, _ = test_img.shape

You will need to change the IP camera source image resolution and that size
will be used. in speed-can,py
Hope this explanation helps.

Slow response was due to a plumbing nightmare that needed attention.

Claude

mlapaglia, 2021-12-29:
Here is the URL in VLC:
[image]

When I look in the UI however, I get a much larger image:
[image]

I'll take a debug strep through the code and see what it's doing, could it possibly be dropping everything after the `?` in the URL?

pageauc, 2021-12-29:
If you want to see the image in original size change config.sys

    image_bigger = 3.0

Also you can reduce the size of the tracking window usint

.
    x_left = 50           # Default=50 comment variable for auto calculate
    x_right = 250         # Defaykt=250 comment variable for auto calculate
    y_upper = 90          # Default=90 comment variable for auto calculate
    y_lower = 150         # Defaykt=150 comment variable for auto calculate

Smaller window will mean faster opencv motion tracking processing

Claude

mlapaglia, 2021-12-30:
i don't think i'm explaining myself well enough.

The image on the web server is 1440p resolution. that means the program is analyzing the 1440p web stream, which takes up a lot of CPU. i want it to analyze the 480p sub-stream that the camera outputs. even though i point the application to `rtsp://username:password@<ip address>/cam/realmonitor?channel=1&subtype=2` it's still analyzing the 1440p stream.

pageauc, 2021-12-30:
If you count the bars you will get by my count 70 that are 10 pixels.  That would be your 704 pixel width image.  The default config.sys has image_bigger = 3.0 but  I think yours might be 2.0. So stream image is resized just for webserver display purposes since it is easier than looking at a small stream size image size on the webserver.  Therefore if you want a different size display image on the web server then adjust the image_bigger accordingly.

The tracking window is too large for your image 704x480 stream image size.  This motion tracking area is cropped and processed for motion tracking so you should size this to best suit the area of motion and reduce opencv processing since smaller is better.   I suggest making it smaller.  eg x_left = 150 and x_right = 550  If you leave the camera pointed as displayed, I would change y_lower to move it up. Since image height is 480 pixels try  y_lower = 200 or 180

I would suggest image_bigger = 2.0  This would double the 704x480 image to 1408x960

Let me know how you make out
Regards Claude ...

closed 2023-11-06 by pageauc