pageauc/speed-camera #107: getting debug to trigger without motion

mlapaglia, 2021-12-28:
is there a way to trigger a screenshot to happen while in debug mode? cars don't drive by my house very often, trying to resize the x,y coordinates for the motion detection is troublesome because of this. A button to grab an image, overlay the red boxes, and show the user what the coordinates would look like would be useful.

pageauc, 2021-12-29:
Matt

In config.sys there are two variables right after calibrate.  Just set
align_cam_on to True and specify the number of seconds you want between
camera shots.
These will be in the same folder as the calibration images.. Should be in
the media folder

calibrate = True         # Create a calibration image file with calibration
> hash markers 10 px per mark
> align_cam_on = False     # Default=False  True Saves alignment image to
> help with camera pointing
> align_delay_sec = 5      # Default=5 seconds delay between each alignment
> image

Let me know if this solves your problem.

Regards Claude ...

closed 2021-12-29 by mlapaglia

mlapaglia, 2021-12-30:
do you mean `config.py` ? I can't find a `config.sys`
When I look in `config.py` I see `calibrate = True` but not `align_cam_on`. when i add it manually it doesn't appear to do anything.

pageauc, 2021-12-30:
my typo yes config.py

I would upgrade to the latest version of speed-camera.  If you are running
on a RPI. then run the menubox.sh  and select the UPGRADE menu option.
After upgrade you will  need to copy the latest config.py.new to config.py
per instructions.
If you are using a docker image on a non RPI then speed camera version may
be out of date.  Someone else set up the docker image and I don't maintain
it.

Claude ...

mlapaglia, 2021-12-31:
ya i'm running the docker image :(

pageauc, 2021-12-31:
The docker image has been update and I have updated the Readme to include
additional Docker Quick Start details
https://github.com/pageauc/speed-camera#docker-install-quick-start
Randall has upgraded the docker image so you should be able to run* docker
build* command

Let me know if this is working for you.
Claude.

mlapaglia, 2022-01-03:
yep it started working for me after the image got updated, ty!