pageauc/speed-camera #109: Feature Request - Tensor Flow Coco SSD / Object Detection

AndreasW79, 2022-01-26:
Hi there,

is there any plan to make object detection optinonal?
Maybe its too much overhead for the cpu (in case of an rpi) but it would be great to recognize the vehicle type (i.e. car, bus, bicycle, people etc.).

Thanks in advance and thanks for your effort !

Best regards,
andreas

pageauc, 2022-01-26:
Only practical way is to write a separate program that could batch process
the images.  Doing it in real time would not be practical unless you had an
add on hardware.
Post processing would be possible to classify the saved images.

pageauc, 2022-01-26:
BTW the ALPR (Automatic license plate reader that is included with speed
camera is an example of post processing.
It reads the database and processes images and updates the DB records as
required.
See code here
https://github.com/pageauc/speed-camera/blob/master/alpr-speed.py

johncblacker, 2023-11-01:
I'm beginning to think I should embrace the concept of recording video and post-processing it on a faster machine - say a p/c or faster SBC if you have such a device.  Performance on the RPi is good but when you have to process several frames in succession it becomes a problem if you can't get the FPS up to a fairly high value; if your FPS value is too low then inaccuracy and missing very fast vehicles.

closed 2024-02-15 by pageauc