pageauc/speed-camera #111: Can't get speed-cam and webserver to work from menubox.sh

Kurisutian, 2022-02-04:
Hi there,

thanks for this awesome project. Got a lot speeding idiots here and this was just what I was looking for. ;-)
However I did get the most recent Raspberry Pi OS (Bullseye) installed on an old RPi3 of mine and wanted to run the substream of my outdoor camera through it to not have a to large image to process on this old device. 
Unfortunately I could not get past the calibration at all since I could not run either one of the programs through the menubox. For the webserver I was able to manually run webserver3.py in the console since the webserver.sh script immediately crashed when started (with no error message whatsoever) and when running the speedcam part it starts but crashes shortly after (perhaps when the first object needs to be detected). So I could not get anything recorded to calibrate at all.
Also I noticed an error message about vcgencmd not being in the /opt/ folder. This I was able to solve pointing the entry in the menubox.sh file to the new location at /usr/bin/vcgencmd

Nevertheless this did not help me at all to get the system calibrated and running which I would love and record as much as possible about the speeding situation here and make it public. So I'm kinda counting on this awesome tool to support this idea. ;-)

Thanks a lot and let me know if I can provide any more data that helps with the fixing.

Kurisutian, 2022-02-15:
I checked again with the recent code. Still, the webserver cannot be started through the menu and the calibration cannot be done as speedcam seems to exit/crash as soon as there is an object passing by. Anything I might be missing when doing my best to do the calibration following the documentation in the wiki? Or might the rtsp stream cause this and it works with a directly connected webcam only?

pageauc, 2022-02-15:
Are you using Debian Bullseye?

Kurisutian, 2022-02-15:
Yes, I do. 
I checked with both variants, 32bit and 64bit as Raspberry OS now supports both as their current version but it did not change anything.

pageauc, 2022-02-15:
raspberry pi camera is not supported on Bulleye due to picamera python
library that is still in development.  I think there are work arounds.
I have not done any testing.
There may also be issues with my webserver as well.

At this time I advise you to install then raspberry pi OS (Legacy) Buster
that will work correctly.  When RPI foundation comes out with a rewrite of
picamera python library I will test on Bulleye.
Regards
Claude ...

Kurisutian, 2022-02-15:
I'm not using a directly connected camera but a rtsp stream of an already existing camera in my network which I can point to the street.
Nevertheless I'll reinstall with buster and see how far I can get with this. At the moment I don't mind an older OS as long as I can get it running. Thanks a lot for the recommendation and your reply.

One question out of curiosity: will the docker version of the setup also run on a Linux system like Open Media Vault since I intend to switch to this replacing my qnap nas. So consuming the compose file in portainer is possible but I don't give anything on running this on a Linux system inside docker. 

Thanks a lot!

Best regards,
Christian

Kurisutian, 2022-02-19:
Setting everything up with buster seems to work (although I do not use the picam at all). Thanks a lot for that hint, now I can start calibrating and positioning the camera so it will detect all vehicles (the first test just seemed to detect the bigger vehicles like busses but ignored all cars).

Thank you very much! :smiley:

closed 2022-02-19 by Kurisutian