pageauc/speed-camera #112: after upgrade speed-cam seems to crash every time a car comes by

johncohnvt, 2022-02-05:
Hi.. I just did an upgrade on speed-camera . No the speed-camera.py seems to crash every time a car goes by.  Eg. If I have the opencv window open, ever time a car comes into view and is recognized, the opencv window abruptly closes and the spaeed_camer.py service needs to be restarted.  If I understand the upgrade I did, it did not overwrite my config.py.. . My old config. says I was at ver 11.20 before the upgrade, the config.py.new installed during upgrade also says ver 11.20 .. which means I was already at ver 11.20 before the upgrade .   Any idea what may be causing this.. or how to debug whats up ?   Thanks !
-jc

pageauc, 2022-02-05:
I have released speed-cam.py ver 11.23.  The problem may be due to a revised version of user_motion_code.py.  You can try deleting/renaming the existing user_motion_code.py and performing an UPGRADE.  Sorry for the delay but Internet was out for a while.  Let me Know Results

johncohnvt, 2022-02-06:
Claude,

  Thanks as always for the quick response. That does seem to have fixed things..  It no longer crashes on each observation.  I am noticing a few other things which are a little off. I notice that    that my attempot to start the speed-camera service in /etc/rc.local  is not working anymore..  I have changed the call there  to
su pi -c "/home/pi/speed-camera/speed-cam.sh start >> /home/pi/sc.log"

and see the following in the log
/home/pi/speed-camera/speed-cam.sh ver 6.00 written by Claude Pageau
-----------------------------------------------
STATUS  - speed-cam.py is Running ...
STATUS  - PID is 854
INFO    - To Stop speed-cam.py execute command below
INFO    - /home/pi/speed-camera/speed-cam.sh stop
Done

I have to start the service manually in ./menubox.sh .. or start it as a service  via
/home/pi/speed-camera/speed-cam.sh start

The other thing I’m noticing is that it appears to be missing some vehicles as they traverse the field when I watch using the OpenCV screen. I think it may be due to some mounded snow near the edge of the viewing box.. might be confusing the image recognition.

Anyway.. things are looking pretty good. Thanks so much for this excellent project !
-jc

pageauc, 2022-02-06:
Did you delete/rename user_motion_code.py before the upgrade.  Install
script will not overwrite existing user_motion_code.py so you need to
delete/rename for it to get updated.  I suspect this was the problem you
were having.  New user_motion_code.py has more parameters passed to the
function so I added a try except TypeError so it would not crash with an
older user_motion_code.py.   This is just for those that want to execute
something after image taken when motion tracking is triggered

BTW using the opencv window will slow processing loop down somewhat

Claude ...

johncohnvt, 2022-02-06:
Yes.. I renamed both the .py and .pyc  for user_ motion _code . That seemed to fix the crash

I turned off the preview once the crash was debuged.. I womder if the slowdown was what was making it miss a few cars.. all seems good now with the remaining mytery of why its not staying running after bring up at boot

<http://www.youtube.com/user/pageaucp>

pageauc, 2022-02-06:
I have found sometimes vehicles are hard to track.  Eg white vehicle with
snow background, green with green background, Birds, squirrels, pets moving
around Etc. setting min area can sometimes help.  I have trees that are in
the tracking window so if it is windy they can cause tracking issues.
There are several config.py timeouts in the loop to avoid tracking problems
like a few movements due to trees that do not result in a completed track.
These get timed out but can also mean a legitimate movement gets missed due
to the prior delay.  Also remember that sometimes contours can move around
due to reflection, lighting, Etc.

Regards Claude ..

closed 2023-11-06 by pageauc