pageauc/speed-camera #113: Python version needs to be specified

oxivanisher, 2022-04-16:
In the newest debian release (11, bullseye), there is no more `python` command available by default. See [this note on the official Debian Wiki](https://wiki.debian.org/Python#Python_in_Debian). Since python 2 is EOL since more than two years now, the correct way would be to default to python 3. I understand that some people will have trouble with this when the OS version was not upgraded, but form a security stand point, this should be done anyways.

So, on newly installed or upgraded raspberry pi's, this project currently is not working out of the box. The following files should be updated:
```
$ grep -R "/usr/bin/env python" *
alpr-speed.py:#!/usr/bin/env python
makehtml.py:#!/usr/bin/env python
search-speed.py:#!/usr/bin/env python
speed-cam.py:#!/usr/bin/env python
sql-make-graph-count-totals.py:#!/usr/bin/env python
sql-make-graph-speed-ave.py:#!/usr/bin/env python
sql_speed_gt.py:#!/usr/bin/env python
```

It is possible to workaround this issue by installing the `python-is-python3` package, but its only a hacky workaround.

The start scripts currently hide the error for the user, so they don't know why its not starting. But if somebody tries to run the python code directly, the error is:
```
$ ./speed-cam.py
/usr/bin/env: ‘python’: No such file or directory
```
I hope this helps more people with problem find this post.

pageauc, 2022-04-16:
Tried a recent and updated Bullseye on a RPI.  ran
*sudo apt install python-is-python3*
says already installed. Trying to execute python-is-python3 says file not
found
find -name only finds a directory at* /usr/share/doc/python-is-python3*

executing *env python*  brings up python3 command shell
So it would appear that as of recently env python works on RPI.  I would
have to install a Debian image and do some testing
but it would appear you have already done some testing.

Bullseye appears to be a bit of a train wreck on RPI as far as the
picamera python library is concerned.  Still waiting on
replacement and not sure what I will do for backward compatibility
for picamera module.

Read the debian wiki link you provided and not not much of implementation
details.

Can you provide a sample python script or implementation script that
implements a solution that is backward compatible..
or detailed instructions that can be added to the speed-camera wiki and
main page.
Thanks

oxivanisher, 2022-04-27:
The workaround for the script with `python-is-python3` works for me also. But as the debian (and RPI OS is based on debian) state in the document, it's probably not going to be available always/forever, since everything should migrate to python3. But this is no real problem, since it would be very easy to change the .sh files to support python2 and python3.

I also tried to get more information about the picam lib and it seems, that they [migrate completely to libcamera](https://www.raspberrypi.com/news/bullseye-camera-system/). So it will probably require some work but unfortunately I have no experience with either libraries or using a camera with python. :( Hopefully the link with the blog entry can give you some help...

pageauc, 2022-10-11:
I am thinking of writing a new libcamera-only version based on existing
code rather than try to integrate picamera and libcamera in one program..