pageauc/speed-camera #114: Log ERROR

slowtoaccept, 2022-06-15:
Log file entry says delete all files in /home/pi/speed-camera/media/recent. 

I'm missing cars (unfortunately, fast cars!) and wonder if this is the problem. 

Also previous error: ERROR    deleteOldFiles Possibly symlink destination File Does Not Exist
Is there a solution that doesn't require manual intervention?
Running picam on a RASP 4B.

Thought this was funny.
[image]

pageauc, 2022-06-15:
To clear up recent folder issues just delete all entries in the recent
folder.

cd media/recent
rm *

This will delete all the symlinks but as more data comes in newer ones will
be created up to the specified max setting in config.py

https://github.com/pageauc/speed-camera/wiki/How-to-Tune-Motion-Tracking-Settings
See this wiki.  You may have to adjust the timeout and x_diff settings per
link above.

Claude ...

slowtoaccept, 2022-06-15:
As a total neophyte to other neophytes, I recommend tutorials on openCV tracking. See for example https://www.youtube.com/watch?v=MkcUgPhOlP8
Perhaps it will explain why a runner is tagged a 56.5 mph.

slowtoaccept, 2022-06-15:
Also, try this alternative: https://core-electronics.com.au/guides/detect-speed-raspberry-pi/

closed 2022-07-01 by slowtoaccept