pageauc/speed-camera #115: Speed-camera on Debian 11/RaspberryPi OS aarch64

will-code-for-pizza, 2022-06-19:
Hi all,
is there a possibility to run speed-camera on Debian 11 aarch64 ??

After successful install with speed-install.sh and some changes in config.py, I debugged my problem with

```
$ ./speed-cam.py
----------------------------------------------------------------------
speed-cam.py 11.23  written by Claude Pageau
Motion Track Largest Moving Object and Calculate Speed per Calibration.
----------------------------------------------------------------------
Loading  Wait ...
Traceback (most recent call last):
  File "/home/wartung/speed-camera/./speed-cam.py", line 339, in <module>
    from picamera.array import PiRGBArray
  File "/usr/lib/python3/dist-packages/picamera/__init__.py", line 72, in <module>
    from picamera.exc import (
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 41, in <module>
    import picamera.mmal as mmal
  File "/usr/lib/python3/dist-packages/picamera/mmal.py", line 49, in <module>
    _lib = ct.CDLL('libmmal.so')
  File "/usr/lib/python3.9/ctypes/__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libmmal.so: cannot open shared object file: No such file or directory

```
The libmmal.so error seems to be related to the non 64-bit functional picamera driver

https://github.com/raspberrypi/firmware/issues/1558#issuecomment-983483179

Under 64-bit 'libcamera' is recommended, but there is also a new version of 'picam' for 10 days, I will try out...
https://forums.raspberrypi.com/viewtopic.php?t=335832

Regards

pageauc, 2022-06-19:
> raspberrypi/firmware#1558 (comment)

You might want to try changing config.py configuration.

Make sure there is NO Webcam plugged in.  Raspberry Pi camera should be
recognized as video device zero and opencv should
be able to read the picamera without loading the picamer python library.
Not sure about mmal but I did a test on my RPI using Buster 32bit
and it ran OK under the WEBCAM setting and not using python picamera
libraries.  Give it a try and let me know.
Claude

*config.py settings *
WEBCAM = True
WEBCAM_SRC = 0

pageauc, 2022-06-19:
Also if this works and image is flipped, physically rotate the picamera
module or RPI 180 deg as appropriate.  I have cases with hole for picamera
as part of the case.  Don't use the USB flip setting due to slow performance

will-code-for-pizza, 2022-06-19:
After a lot of tests, I can say, that the issue is related to Debian 11 and Python3 -  also with 64 bit architecture.
Works like charm under Debian 10 32 bit

The only error I could find, was during installation script:

```
Package python3-imaging is not available but is provided by another package
referenced. This can mean that the package is missing, that it has been detached
or is only available from another source.
However, the following packages replace it:
   python3-pil

E: There is no installation candidate for package 'python3-imaging'.

```

will-code-for-pizza, 2022-06-19:
Sorry, I have to update this...

```
$ uname -a
Linux speedcam 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux

```
```

$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

```
$ ./speed-cam.py
----------------------------------------------------------------------
speed-cam.py 11.23  written by Claude Pageau
Motion Track Largest Moving Object and Calculate Speed per Calibration.
----------------------------------------------------------------------
Loading  Wait ...
----------------------------------------------------------------------
Note: To Send Full Output to File Use command
python -u ./speed-cam.py | tee -a log.txt
Set log_data_to_file=True to Send speed_Data to CSV File speed-cam.log
----------------------------------------------------------------------

Debug Messages .. verbose=True  display_fps=True calibrate=True
                  show_out_range=True
Plugins ......... pluginEnable=True  pluginName=picam240
Calibration ..... cal_obj_px_L2R=80 px  cal_obj_mm_L2R=4700 mm  speed_conv_L2R=0.21150
                  cal_obj_px_R2L=85 px  cal_obj_mm_R2L=4700 mm  speed_conv_R2L=0.19906
                  (Change Settings in /home/wartung/speed-camera/plugins/picam240.py)
Logging ......... Log_data_to_CSV=True  log_filename=speed-cam.csv (CSV format)
                  loggingToFile=True  logFilePath=speed-cam.log
                  SQLITE3 DB_PATH=/home/wartung/speed-camera/data/speed_cam.db  DB_TABLE=speed
Speed Trigger ... Log only if max_speed_over > 0 kph
                  and track_counter >= 6 consecutive motion events
Exclude Events .. If  x_diff_min < 1 or x_diff_max > 30 px
                  If  y_upper < 60 or y_lower > 17 px
                  or  x_left < 20 or x_right > 300 px
                  If  max_speed_over < 0 kph
                  If  event_timeout > 0.40 seconds Start New Track
                  track_timeout=0.00 sec wait after Track Ends (avoid retrack of same object)
Speed Photo ..... Size=960x720 px  image_bigger=3.0  rotation=0  VFlip=True  HFlip=True 
                  image_path=media/images  image_Prefix=speed-
                  image_font_size=12 px high  image_text_bottom=True
                  image_jpeg_quality=75  image_jpeg_optimize=True
Motion Settings . Size=320x240 px  px_to_kph_L2R=0.211500  px_to_kph_R2L=0.199059 speed_units=kph
                  CAM_LOCATION= None
OpenCV Settings . MIN_AREA=100 sq-px  BLUR_SIZE=10  THRESHOLD_SENSITIVITY=20  CIRCLE_SIZE=5 px
                  WINDOW_BIGGER=1 gui_window_on=False (Display OpenCV Status Windows on GUI Desktop)
                  CAMERA_FRAMERATE=30 fps video stream speed
Sub-Directories . imageSubDirMaxHours=0 (0=off)  imageSubDirMaxFiles=2000 (0=off)
                  imageRecentDir=media/recent imageRecentMax=300 (0=off)
Disk Space  ..... Disabled - spaceTimerHrs=0  Manage Target Free Disk Space. Delete Oldest jpg Files
                  spaceTimerHrs=0 (0=Off) Target spaceFreeMB=1000 (min=100 MB)

----------------------------------------------------------------------
Logging to File speed-cam.log (Console Messages Disabled)
OpenCV Error: Assertion failed (dims <= 2 && step[0] > 0) in locateROI, file /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp, line 949
Traceback (most recent call last):
  File "./speed-cam.py", line 2025, in <module>
    speed_camera()  # run main speed camera processing loop
  File "./speed-cam.py", line 1347, in speed_camera
    grayimage1, contours = speed_get_contours(image2, grayimage1)
  File "./speed-cam.py", line 1216, in speed_get_contours
    differenceimage = cv2.blur(differenceimage, (BLUR_SIZE, BLUR_SIZE))
cv2.error: /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp:949: error: (-215) dims <= 2 && step[0] > 0 in function locateROI
```

will-code-for-pizza, 2022-06-19:
> WEBCAM = True 
> WEBCAM_SRC = 0

These settings have no affect to the error

pageauc, 2022-06-19:
Will

My suggestion would be to run under Raspberry Pi OS Legacy Buster.  Speed
camera works OK on that distro.  Currently Bullseye is not supported since
the original picamera python modules are Not supported on Bullseye and
other distro issues. The new picamera2 module is still under development
with no definitive date for production.  Also as you have found out a lot
of libraries have changed and python 2 is no longer supported.  I will not
be looking at these issues until the distro matures and I have the time and
inclination to review.  I have been retired for 16 years and did this
project just as a fun coding project.
If you want to clone the project and develop your own solutions that
support backward and forward compatibility I would be more than happy to
review and merge any github pushes.

Regards
Claude ..

Claude ...

will-code-for-pizza, 2022-06-20:
Hi Claude, thanks for that statement.

The issue is, that I currently use Raspberry Pi OS Legacy Buster.
Yesterday I had serveral installations (from Raspberry Pi OS Lite Bullseye 64 bit down to my current installation Raspberry Pi OS Legacy Buster) and obviously, on Legacy Buster the installation process runs much better.

I have to mention, that my user accout is NOT 'pi', thats why I opened another ticket with "improvements".
I will do a totally fresh install of Raspberry Pi OS Legacy Buster, the 'pi' user and speed-cam and would document the remaining problems here.

Don't worry about... I know, this is a hobby project. I just want to give you some informations to improve some issues.
Regards
Will

pageauc, 2022-06-20:
Yes I am aware of the issues new distro's can cause as well as what
standard packages and libraries are included/restructured/renamed/ Etc.
This has happened before and can cause downstream issues for existing
software since they often don't provide backward compatibility.  A simple
parameter change can often cascade and cause a lot of trouble shooting.
System and Python Libraries are usually the main cause.This has happened in
the past but seems to be a bigger issue with Bullseye.  Python is also
going through a transition. At some point I will determine whether I will
try to maintain backward compatibility or create a new version for the
latest distro.  In the past I have tried to maintain python2 and 3
compatibility as well as changes in distro support packages that change eg
ffmpeg
Claude ...