pageauc/speed-camera #123: How to use a USB camera instead of picamera

johncblacker, 2022-12-04:
I'd like to use a usb camera (logitech c270) instead of the picamera.  How do I configure it?  Do I need to specify a device number/index or anything?

pageauc, 2022-12-05:
Edit config.py webcam_on = True
If you do not have any other camera installed including camera module then
set wecam_src = 0
This should be default. If you have a picamera it will take 0 so you must
likely have to set to = 1

See comments in config.py.  rtsp protocol network cameras are also xupported

pageauc, 2022-12-05:
Replied while not near my computer.
Settings to change in config.py   nano configl.py

WEBCAM = True

This should work if you only have a webcam plugged in and not other
camera.  Otherwise see previous email.
You should not have to change the other WEBCAM settings to get started.
Defaults should work OK.
I have tried speed camera with logitech web cam and had no problems.

Let me know if you have any issues.
Regards Claude ...

johncblacker, 2022-12-05:
OK, I'll give that a try.  I did notice, though, that if I had the logitech camera plugged into a USB port and tried to boot my pi, it would not boot.  But, also, since my sd card reader is toast, I am booting my pi via USB so perhaps that's the problem; although my keyboard and mouse don't seem to both the boot process...

johncblacker, 2022-12-05:
Well, it looks like I can't get anything to work with bullseye.  Enabled legacy camera support and couldn't access the camera.  And, as I mentioned above, if you're booting from a USB drive you can't have a Logitech c270 plugged into a USB port at boot time because it won't boot.  So, I'm going back to the "legacy buster system" that was "so kindly" made available by the Raspberry Pi folks.  I'd be interested in knowing how you got the legacy cam to work with bullseye.  when I do a vcgencmd get_camera with legacy support, I see: "supported=1 detected=1, libcamera interfaces=0" and when I try to start speed-cam.py it fails get find the camera and won't start.  So, back to buster for me.  They can stick buster wherever they want.

pageauc, 2022-12-05:
You should not be seeing problems booting with USB camera plugged in.
Sounds like you may have a defective SD card or boot image.
You might want to try using Raspberry Pi Imager and create a new OS image.
Maybe try another SD card.  Whenever strange things happen it usually gets
resolved by just creating a new image.  Make sure your power supply is
adequate.  This can also cause issues if there is not enough power to
support USB camera power demands.

pageauc, 2022-12-05:
Just enabled legacy pi camera using sudo raspi-config,  Interfaces and
enabled legacy.  Did not have any problems booting or running pi-camera
under Bulleyes.  I currently have several RPI's running Bullseye in
legacy.  Make sure to reboot when raspi-config prompts you or manually
reboot to enable legacy mode.  Also see my comment about possible corrupt
SD card or inadequate power supply to power USB camera and RPI

closed 2023-11-06 by pageauc