pageauc/speed-camera #124: howto improve the boudingbox around the vehicle ?

sebsoft, 2022-12-21:
Hi Claude,
Love your work.  I'm trying to use speed-camera to estimate the size of the passing vehicles into different classes like, car, van and truck.
The picture below shows a  vehicle passing the road in front of my house. I have a perfect 90 degrees view.   It seems that the height of the bounding box is most of the times correct but the length is usually incorrect.  I could estimate the vehicle class using the heigth only but are there any settings which will get the box around the complete vehicle ?

[image]

[image]

pageauc, 2022-12-21:
Speed camera is an object detection rather than vehicle detection.  It
finds the largest moving contours and tracks motion while filtering out
data points that are not close to the previous motion track event or times
out.  Eg motion stops before completing the track.

Most often a vehicle will have certain areas that are picked up as a
separate contour.  Eg wheel well, truck, door Etc.  If the object is far
enough away it can lock onto the whole object depending on the background.
Eg white vehicle with snow background or green vehicle and green
background.  It is possible to do AI object detection using AI or haar
cascade method.  This takes quite a bit of processing.  Speed camera does
not use object detection. It uses only motion detection for tracking.

I have been thinking of doing post processing on speed images to do object
detection
Here are some links fyi
AI deep learning example https://github.com/MaryamBoneh/Vehicle-Detection
Haar Cascade example
https://github.com/andrewssobral/vehicle_detection_haarcascades

Speed camera Object Tracking was written to be a simple real time project
that I wrote for fun and at the time the Raspberry Pi was very limited in
processing power.
It would be possible to write something using deep learning  object
recognition if someone would like to take on the project.

Let me know if this answers your question.
Claude ...

sebsoft, 2022-12-28:
Hi Claude,

Thank you for elaborate answer.  Post processing the captured images would indeed be a elegant solution. Let see how far i get with that.

Sebastiaan

sebsoft, 2023-02-01:
Hi Claude

I did a post processing test on the desktop with yolo which has a pretrained model with several object classes which includes cars, trucks bikes etc.  First results look pretty good.  The red box on the picture with id=2 is the result of the object detection.
Have to check if it runs on the rpi as well.

snippet

       net = cv2.dnn.readNet("models/yolov4.weights", "models/yolov4.cfg")
        self.model = cv2.dnn_DetectionModel(net)
        self.model.setInputParams(size=(832, 832), scale=1 / 255)
        # Allow classes containing Vehicles only
        self.classes_allowed = [0,1, 2, 3, 5, 6, 7, 8, 9]
        #person
        #bicycle
        #car
        #motorbike
        #aeroplane
        #bus
        #train
        #truck
        #boat

[image]

pageauc, 2023-02-01:
Thanks for the update.  If you decide to post code online, a link would be
appreciated.
Claude ...

SamDecrock, 2024-03-23:
If I read the code correctly, it saves the last image with its found rectangle. @sebsoft to improve the rectangle, you could keep track of all images for a certain vehicle detection event and save the one with the biggest rectangle.

Mobilitysensing, 2024-03-25:
See below the result of another approach.  The first and last image are saved to disk and together with the timespan between them passed to the code which does the object detection.  Below the last image with the object detection box from the first image added.  The blue line connects the centers of the the boxes.  Using the pixel / m values the length of the object is calculated and from the length of the blue line the speed of the object. 

[image]

a113speederbot, 2024-03-25:
> See below the result of another approach. The first and last image are saved to disk and together with the timespan between them passed to the code which does the object detection. Below the last image with the object detection box from the first image added. The blue line connects the centers of the the boxes. Using the pixel / m values the length of the object is calculated and from the length of the blue line the speed of the object.
> 
> ![Stop_20240322-1546098_car_4 9_41_](https://private-user-images.githubusercontent.com/111275105/316424521-6ca69912-d8f0-484e-8485-dbf6bd9c8a3d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEzNzI0MzgsIm5iZiI6MTcxMTM3MjEzOCwicGF0aCI6Ii8xMTEyNzUxMDUvMzE2NDI0NTIxLTZjYTY5OTEyLWQ4ZjAtNDg0ZS04NDg1LWRiZjZiZDljOGEzZC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMyNVQxMzA4NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04YjNiZDBkYzE4MDYyZTcwNGZkZWRjMzRkMmU3ODU0ZGZkYjdiZjYyZTc1NmM1NzM4OGI1ZTJhNjQ0ZWVjMzhjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.LiwzJhG4ya-DqifyYd-My7NvL_5v7xA_DqQQOltS7mA)

Wow, that's clever - I like that! Well done.

sebsoft, 2024-03-26:
> @Mobilitysensing Thats very nice! Can you share the code changes you did for this? This will help me tweak the parameters a bit better. I still see cars passing by at unrealistic speeds in my setup.

You can find the complete code here. ( It needs some polishing as it is much based on trial and error)
It's based on motrack by @pageauc  which only contains the core of speed-camera.  There are a few hard coded numbers which need to be changed to get results with a different camera setup.  I was only interested in cars on the road passing from R to L so these are filtered out in motrack.py . Only cars passing R to L are passed on to user_code_speed.py where the object detection happens.  As the object detection will detect all objects in the image there are some constants to determine weter the object is on the road.

It works pretty well but depends on lightning conditions.  When cars cast shades on the road they can sometimes be classified as other object (boat, plane etc) and the boundingbox includes the shade.

https://github.com/sebsoft/motrack