pageauc/speed-camera #125: Upload speed image to server via FTP

waubain, 2022-12-22:
I am new to Python(I work mostly in PHP). I can upload a file saved in the images folder via FTP to my web server using this code.

```
import ftplib
import os

FTP_HOST = "myserver.com"
FTP_USER = "username"
FTP_PASS = "password"
from_path = 'speed-camera/media/images/'
filename = 'speed-20221222-131530.jpg' 

ftp = ftplib.FTP(FTP_HOST, FTP_USER, FTP_PASS)

ftp.cwd('public_html/speed-images/')

image_file = open(os.path.join(from_path, filename), 'rb')
ftp.storbinary('STOR {filename}', image_file)

image_file.close
ftp.quit()
```

I would like to have the image saved directly to my web server in addition to the images folder. It is unclear to me how I would go about doing this in speed-cam.py. 

Any hints would be appreciated.

pageauc, 2022-12-23:
You should add your code to user_motion_code.py.  See comments in code.

https://github.com/pageauc/speed-camera/blob/master/user_motion_code.py

Various variables are passed to the function including the filenamePath.
You would need to extract the filename without the path

per filenameOnly = os.path.basename(filenamePath)

user_motion_code.py will automatically get executed when a speed camera is
triggered.

Let me know if this answers your question
Merry Christmas
Claude ...

pageauc, 2022-12-23:
FYI rclone also supports FTP.  Speed camera installs rclone python library
by default.  For details see

https://rclone.org/ftp/

Also see speed camera sample shell scripts and the speed camera wiki entry
Claude ..

waubain, 2022-12-24:
I stayed with your first option. The code was the most logical to me. We
are having blizzard conditions here in the Midwest with cold and high
winds. No one is out driving (a good thing). I had to wait awhile for a car
to go by to test it out.
Thank you for the information. Have a Merry Christmas and a great New Year.

Bob

pageauc, 2022-12-27:
Hi Bob.

I would be interested in what web server type/ver and the speed code base you are using for the web server eg HTML, CSS, Java, etc
Would it be possible to zip an example or supply link and send to me.  I can give you my email address
Have a Happy New Year
Claude ...

waubain, 2022-12-28:
Hi Claude,
I use a hosting site with a shared Linux server. I am running the speed
code on a PC with Windows 10. I use an IP camera mounted above my garage
door. I think it is a little too far from the road
I am a retired hospital pharmacist, with an interest in agriculture and
weather. I have a website displaying data, combining data from both of
those. My programming skills are beginner level and mostly everything I do
is from manipulating examples I find on the web.
My site is HTML, CSS, PHP and some Javascript (of the plain vanilla type) .
All data is stored on a MySQL database on the hosted server.

I am going to store the images from the "speed-camera" program on my site,
where only I can see, but the speed data, will be stored in a table and
displayed on a web page similar to the images page on my site.

If you send me your email address I can send you the URL to my site. If you
see something of interest I can send you a zip file.

pageauc, 2022-12-28:
Everyone learns from others and most developers build a certain amount from
other open source code or examples.  Libraries a just shared code with good
API structures for others to use.  I like you FTP example.

Later on you may get an idea of your own and develop your own solution.
That is fun for me.

My Email address is

***@***.***

zip file would be appreciated.

Thanks Claude ...

waubain, 2022-12-29:
Claude,
Github seems to be encrypting all the email addresses. Can you send me a
message through my profile at GitHub?
Bob

closed 2022-12-30 by waubain