pageauc/speed-camera #127: x64 Bullseye compatibility issues

cluelesshack, 2023-01-08:
Claude, thanks for sharing your project. Is this compatible with x64 Bullseye? I see it uses the legacy camera interface, but it doesn't seem like x64 Bullseye is compatible with anything but libcamera (even though the option exists)? If it matters, I have open cv 4.7 installed as well. Currently, speed-cam.sh appears to run, but I don't see any images. If I launch it via terminal, it runs and provides a PIV, but I don't see any other activity. When I run status, it says that it isn't running. Not sure if the camera discrepancy is the root cause or not. Appreciate any help.

pageauc, 2023-01-24:
I have not tested with 64 bit Bullseye.  Only 32 Bit Bullseye in Legacy mode under python 3.

I wrote a small demo/test repo for testing threaded camera streaming using class imports for picam2, picam legacy, webcam and IP camera.

    https://github.com/pageauc/MoTrack-Picam2-Demo

This seems to work but I am still not happy with it and will continue testing.  I do not have an install script but if you have already installed speed-camera then there  should not be any issues.  Just clone repo. I am thinking of creating a new speed camera repo that ditches pi camera legacy.  Anyway it is a work in progress as long as my wife does not schedule too many things.  BTW we have been married for going on 53 years so Happy Wife, Happy Life

Claude ...

sebsoft, 2023-02-01:
Hi Claude 

Possibly related. I just got the new v3 camera which only seems to work with libcamera.  Would it be possible to use the v3 camera with speed-camera ?

All the best for you and your wife
Sebastiaan

pageauc, 2023-02-01:
Not sure if picamera2 python lib supports v3 yet.  Speed camera currently
does not support libcamera but I have test code at
https://github.com/pageauc/MoTrack-Picam2-Demo
Still testing...

sebsoft, 2023-02-17:
Hi Claude,

I've tested your motrack demo code and can confirm that the new v3 camera works well.  In my use case cars are being detected properly at 640 x 480.

I've added your streampilibcam to speed-camera and got it working. It seems however that the framerate is much lower here.  Pedestrians are detected but cars drive to fast.  Any ideas to speed it up ?

Best Regards
Sebastiaan

pageauc, 2023-02-17:
Yes picamer2 libcam library is stated to be slower but code is still in
dev.  If FPS are lower you can try setting config.py motion event setting
*track_counter* lower. default is 6.  Try ramping down.  Also Under
Bullseye you can always go back to legacy picamera using sudo raspi-config
under Interface Options and reboot.  This will revert back to original
picamera python library

pageauc, 2023-02-21:
I did some testing and changed https://github.com/pageauc/MoTrack-Picam2-Demo/blob/master/streampilibcam.py
to put capture in update.  This speeds things up.  Can you give it a try and let me know your results.  I am planning on updating speed camera.  Claude

sebsoft, 2023-02-22:
Hi Claude,

This seems to solve the problem.  Speed camera now works like expected with  track_counter set to 6 and using 640 x 480 resolution.  Car's drive about 50 km/h on the road the camera is pointing at.  I'll do some more tweaking with the settings and do a proper calibration tomorrow.   It would be nice if speed-camera gets updated, the new V3 camera is not supported by the legacy picamera.

Before you tweaked streamlibcam.py i already got good results with MoTrack but not with speed-camera.  Does this mean that speed-camera has parts in the code which slow down the effective framerate in the end ?

Sample picture below using the V3 camera , low light conditions and not yet calibrated.

[image]

Sebastiaan

pageauc, 2023-02-22:
Moved the capture array from read to inside of loop in update.  Speed up is
due to buffering of image array data inside the loop so when a read is
called an image already exists.  I had to put a small delay in loop for
this to work.  I will look at updating speed-cam.py
Thanks for the feedback. Much appreciated
Claude ...

carterw, 2023-04-09:
> I've tested your motrack demo code and can confirm that the new v3 camera works well. In my use case cars are being detected properly at 640 x 480.
>
> I've added your streampilibcam to speed-camera [...]

I'm running bullseye with the v3 camera and was hoping to get this to work. I ran the motrack demo code and it does detect motion. What would I do in order to add "streampilibcam to speed-camera and got it working"?

pageauc, 2023-04-09:
I have a working version but have not posted since changes needed for
installation script and testing.  I did test on 64 bit bullseye on libre le
potato and worked successfully. Also RTSP and various usb cams and no
issues. Also works  fine on various RPI hardware but don't have V3 RPI
Cam.  Will ditch python 2 support. Was hoping to add option to read from a
video file or from directory glob spec.

FYI. I severely damaged fingers on left hand in a saw accident and still
recovering.  Doc days 6 months to a year for recovery.
Will try to publish new version in a week or so.
Claude ...

carterw, 2023-04-09:
Wow, sorry to hear about the fingers, I hope you are right-handed. I will be on the lookout for a new version.

sebsoft, 2023-04-11:
Hi Claude

I've made some progress with object detection and speed calculation based on detected objects.  I'am using tensorflow lite with a pretrained model.  It takes about 0.6 seconds to do object detection on an rpi. With the V3 camera this seems to be fast enough to do in the main loop. (and could be used instead of the track_timeout )

I' am saving an image on the start track event together with an image that is saved on the track end event.  Then i do object detection for both images and draw the returned boxes on the end event image.  The distance between the centerpoints of both boxes is then used to calculate distance and speed. From the size of the box the dimensions of the car can be calculated, the object detection add the classification.  In the sample below its visualized together with the trackpoints.  From the first results it looks like the speed can be determined more accurate from the boundingboxes.

I've integrated the code with your motrack sample as this is nice and clean for me learning python.  Right now the code has some specific lines for the road in front of my house as the object detection will detect parked cars and other things on the image ass well.  I'll try to generalize and then publish the code.  

Also made a first attempt to adapt the webserver to work without frames and use a css file.  The "image_bigger" feature is now done inside the browser which saves diskspace on the rpi.

Some samples below. Last on has a copy / paste of the car on the start image to end image.

[image]

[image]

[image]

pageauc, 2023-04-11:
Looks good.  Will you be publishing your version on your own github repo?

carterw, 2023-04-12:
I was looking at the Picamera2 Library documentation and there is an Asynchronous capture mode documented in section 6.5 which may improve performance. There's some example code on page 57. It looks like you could call capture_array() in a loop that will not block on the capture, and pass in a function that would be called when the capture is completed.

It might eliminate the need for a thread in PiLibCamStream . The asynchronous function would simply queue up image arrays in a FIFO as they are generated , and when read() is called you would pop off one or more image arrays to return.

sebsoft, 2023-04-14:
> Looks good. Will you be publishing your version on your own github repo?
> […](#)
> -- YouTube Channel at https://www.youtube.com/user/pageaucp <http://www.youtube.com/user/pageaucp> GitHub Repository at https://github.com/pageauc

Hi Claude,

Work in progress can be found here

https://github.com/sebsoft/motrack

S.

carterw, 2023-05-08:
Thanks! I had made a working prototype for myself, but will download your new version and try it out.

"reading from video file" would be an extremely helpful feature for making calibration easier. A video file containing a recording of a car passing by at a known speed could be repeatedly evaluated with different calibration numbers until the correct values were reached. Or even easier, the known speed could be an input parameter and the code could infer and print the correct values automatically.

jonas18z, 2023-06-21:
I have the v3 camera module and updated everything tried both x32 and x64 version of rpi os. Is there something i need to change to setup.py to get the v3 camera to work? 

`Loading Wait...
----------------------------------------------------------------------
speed-cam.py 12.00  written by Claude Pageau
Motion Track Largest Moving Object and Calculate Speed per Calibration.
----------------------------------------------------------------------
2023-06-21 23:39:33 INFO     validate_cam Connecting to camera pilibcam
2023-06-21 23:39:33 INFO     is_pi_legacy_cam Check for Legacy Pi Camera Module with command - vcgencmd get_camera
2023-06-21 23:39:33 ERROR    is_pi_legacy_cam Problem Finding Pi Legacy Camera supported=0 detected=0, libcamera interfaces=0
2023-06-21 23:39:33 ERROR    is_pi_legacy_cam Check if Legacy pi Camera is Enabled and Camera is working.
[0:16:28.071564039] [1795]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[0:16:28.094430910] [1799] ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Device or resource busy
[0:16:28.109075355] [1799]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media0 and ISP device /dev/media1
2023-06-21 23:39:33 INFO     _initialize_camera Initialization successful.
[0:16:28.111658530] [1795]  INFO Camera camera.cpp:841 Pipeline handler in use by another process
2023-06-21 23:39:33 ERROR    __init__   Camera __init__ sequence did not complete.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 242, in __init__
    self._open_camera()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 441, in _open_camera
    raise RuntimeError(f"camera.acquire() returned unexpected code: {acq_code}")
RuntimeError: camera.acquire() returned unexpected code: -16

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jonte/speed-camera/./speed-cam.py", line 1937, in <module>
    vs, cam = create_cam_thread(mycam)
  File "/home/jonte/speed-camera/./speed-cam.py", line 433, in create_cam_thread
    vs = PiLibCam(size=(CAMERA_WIDTH, CAMERA_HEIGHT),
  File "/home/jonte/speed-camera/strmpilibcam.py", line 28, in __init__
    self.picam2 = Picamera2()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 249, in __init__
    raise RuntimeError("Camera __init__ sequence did not complete.")
RuntimeError: Camera __init__ sequence did not complete.`

pageauc, 2023-06-22:
I don't have a v3 camera module so can't help trouble shoot.  It is
overkill for speed-camera.  The libcamera code is pretty basic per
https://github.com/pageauc/speed-camera/blob/master/strmpilibcam.py
The python trace from you per below indicates camera initialization failed
due to pipeline handler in use by another process.  You will need to do
some testing to see what is causing the issue. see
https://forums.raspberrypi.com/viewtopic.php?t=350501
and https://github.com/raspberrypi/picamera2/issues/627  for some help with
your issue

[0:16:28.111658530] [1795] INFO Camera camera.cpp:841 Pipeline handler in
use by another process
2023-06-21 23:39:33 ERROR init Camera init sequence did not complete.

Let me know if you resolve your problem

carterw, 2023-06-22:
> I have the v3 camera module and updated everything tried both x32 and x64 version of rpi os. Is there something i need to change to setup.py to get the v3 camera to work?

You did verify that the camera works when you operate it with other code, right? You can run the examples in the picamera2 repository?
[https://github.com/raspberrypi/picamera2/tree/main/examples](url)

pageauc, 2023-06-22:
As I don't have a v3 camera you can try modifying the code in
https://github.com/pageauc/speed-camera/blob/master/strmpilibcam.py
I used the picamera2 examples as basis for the code.  Not sure why v3 would
not work.

pageauc, 2023-06-22:
> Thanks for all reply!
>
> I think I figure out one problem but I will take it from start.
>
> I started the menubox.sh and from that i started speed_cam.py and
> webserver.py. But why I tried to acess the page I got a black page with
> text.
> [image: Screenshot 2023-06-22 at 12 24 51]
> <https://user-images.githubusercontent.com/2670662/247902773-fd309aef-803c-49b3-be8d-831b15b2a526.png>
>
> I tough that there was some error that the camera was not started so tired
> to start it but than I got the error message in first post that the camera
> was busy. ( I also tried the picamera-hello example so maybe that was
> running in some test)
> I was also testing in the middle of night so there was no vehicles running
> around here so I didn't get an jpeg images in the folder. But today in the
> morning I got images. But still I can't se anything on the webserver just
> the black box with some text.
>
> TL;RD The V3 camera works but not the webinterface (or maybe I need to
> make some configurations)
>
> Also the camera is cropped but that's maybe some setting.

You need to select the images folder on the right pane of the webserver per
instructions on the page.  There are several folders for different
purposes.  See github  wiki https://github.com/pageauc/speed-camera/wiki
for more help.
Claude

jonas18z, 2023-06-22:
Thanks! 

If some else has problem with crop on the v3 camera I added   raw={"size":self.picam2.sensor_resolution} to strmpilibcam.py That will fix the issue and I now change the resolution to 960x400 that will cover more or less only the read for me with full sensor wide isteed of crooped. 
 
`    self.picam2.configure(self.picam2.create_preview_configuration(
                              main={"format": 'XRGB8888',
                              "size": size},raw={"size":self.picam2.sensor_resolution},
                              transform=Transform(vflip=vflip,
                                                  hflip=hflip)))`

closed 2023-11-06 by pageauc