pageauc/speed-camera #128: USB Camera not working

angrytrkr, 2023-02-14:
There seems to be an issue possibly isolated to my inexperience with programming. I for the life of me cannot get the program to run on my usb camera. There is no Pi camera installed just the usb one. No other video devices in v4l2. I have attemted to change the webcam source but do i need to select the device number for it?

pi@raspberrypi:~ $ v4l2-ctl --list-devices
USB 2.0 Camera: USB Camera (usb-xhci-hcd.0.auto-2):
	/dev/video0
	/dev/video1
	/dev/media0

# Calibration Settings
# --------------------
calibrate = True         # Create a calibration image file with calibration hash markers 10 px per mark
align_cam_on = False     # Default=False  True Saves alignment image to help with camera pointing
align_delay_sec = 5      # Default=5 seconds delay between each alignment image

cal_obj_px_L2R = 80      # L2R Moving Objects, Length of a calibration object in pixels
cal_obj_mm_L2R = 4700.0  # L2R Moving Objects, Length of the calibration object in millimetres

cal_obj_px_R2L = 85      # R2L Moving Objects, Length of a calibration object in pixels
cal_obj_mm_R2L = 4700.0  # R2L Moving Objects, Length of the calibration object in millimetres

# Note if tested speed is too low increase appropriate cal_obj_mm  value and redo speed test for desired direction.
# IMPORTANT - If plugins Enabled Edit Settings in specified plugin file located in plugins folder.

# Plugins overlay the config.py variable settings
# -----------------------------------------------
pluginEnable = True
pluginName = "webcam480" # Specify filename in plugins subfolder without .py extension per below
                        # picam240, webcam240 (Recommended for RPI2 or greater)
                        # picam480, webcam480, picam720, webcam720  (can use RPI3 but Test)
                        # picam1080   (Experimental Not Recommended)
                        # secpicam480, secwebcam480 (Experimental no CSV entries)

# Display opencv windows on gui desktop
# gui_window_on suppresses All Windows if False
# ----------------------------------------------
gui_window_on = False  # True= Turn On All desktop GUI openCV windows. False=Don't Show (req'd for SSH) .
gui_show_camera = True # True=Show the camera on gui windows. False=Don't Show (useful for image_sign)
show_thresh_on = False # Display desktop GUI openCV cropped threshold window. True=Show, False=Don't Show
show_crop_on = False   # Same as show_thresh_on but in color. True=Show, False=Don't Show (Default)

# Display and Log settings
# ------------------------
verbose = True         # True= Display basic status information on console False= Off
display_fps = False    # True= Show average frame count every 1000 loops False= Off
log_data_to_CSV = False # True= Save log data as CSV comma separated values  False= Off
loggingToFile = False  # True= Send logging to file False= No Logging to File
logFilePath = 'speed-cam.log'  # Location of log file when loggingToFile=True

# Motion Event Settings
# ---------------------
SPEED_MPH = True       # Set Speed Units   kph=False  mph=True
track_counter = 6      # Default= 6 Number of Consecutive Motion Events to trigger speed photo. Adjust to suit.
                       # Suggest single core cpu=4-7 quad core=8-15 but adjust to smooth erratic readings due to contour jumps
MIN_AREA = 200         # Default= 200 Exclude all contours less than or equal to this sq-px Area
show_out_range = True  # Default= True Show Out of Range Events per x_diff settings below False= Off
x_diff_max = 24        # Default= 20 Exclude if max px away >= last motion event x position
x_diff_min = 1         # Default= 1 Exclude if min px away <= last event x position
x_buf_adjust = 10      # Default= 10 Divides motion Rect x for L&R Buffer Space to Ensure contours are in
track_timeout = 0.5    # Default= 0.5 Optional seconds to wait after track End (Avoids dual tracking)
event_timeout = 0.3    # Default= 0.3 seconds to wait for next motion event before starting new track
max_speed_over = 00     # Exclude track if Speed less than or equal to value specified 0=All
                       # Can be useful to exclude pedestrians and/or bikes, Etc or track only fast objects

# Motion Tracking Window Crop Area Settings
# -----------------------------------------
# Note: Values based on 320x240 image stream size.
# If variable is commented, value will be set automatically based on image size.
# To see motion tracking crop area on images, Set variable image_show_motion_area = True
# Set align_cam_on = True to help with adjusting settings.
x_left = 50           # Default=50 comment variable for auto calculate
x_right = 250         # Default=250 comment variable for auto calculate
y_upper = 90          # Default=90 comment variable for auto calculate
y_lower = 150         # Default=150 comment variable for auto calculate

# Camera Settings
# ---------------
CAM_LOCATION = 'None'  # Specify an address, physical location Etc for camera
WEBCAM = True          # Default= False False=PiCamera True= USB Webcam or RTSP,IP Camera

# Web Camera Settings
WEBCAM_SRC = 3         # Default= 0   USB camera device connection number
                       # or RTSP cam string eg "rtsp://192.168.1.101/RtspTranslator.12/camera"
                       # see WIKI for details
WEBCAM_WIDTH = 1920    # Default= 320 USB Webcam Image width ignored for RTSP cam
WEBCAM_HEIGHT = 1080   # Default= 240 USB Webcam Image height ignored for RTSP cam
WEBCAM_HFLIP = False   # Default= False USB Webcam flip image horizontally
WEBCAM_VFLIP = False   # Default= False USB Webcam flip image vertically
                       # IMPORTANT Webcam Streaming Performance Hit if Stream Flipped.

# Pi Camera Settings
# ------------------
CAMERA_WIDTH = 320     # Image stream width for opencv motion scanning Default=320
CAMERA_HEIGHT = 240    # Image stream height for opencv motion scanning  Default=240
CAMERA_FRAMERATE = 22  # Default= 20 Frame rate for video stream V2 picam can be higher
CAMERA_ROTATION = 0    # Rotate camera image valid values are 0, 90, 180, 270
CAMERA_VFLIP = True    # Flip the camera image vertically if required
CAMERA_HFLIP = True    # Flip the camera image horizontally if required

# Camera Image Settings
# ---------------------
image_path = "media/images"   # folder name to store images
image_prefix = "speed-"       # image name prefix
image_format = ".jpg"         # Default = ".jpg"  image Formats .jpg .jpeg .png .gif .bmp
image_jpeg_quality = 98       # Set the quality of the jpeg. Default = 95 https://docs.opencv.org/3.4/d8/d6a/group__imgcodecs__flags.html#ga292d81be8d76901bff7988d18d2b42ac
image_jpeg_optimize = True    # Optimize the image. Default = False https://docs.opencv.org/3.4/d8/d6a/group__imgcodecs__flags.html#ga292d81be8d76901bff7988d18d2b42ac
image_show_motion_area = True # True= Display motion detection rectangle area on saved images
image_filename_speed = True   # True= Include speed value in filename
image_text_on = True          # True= Show Text on speed images   False= No Text on images
image_text_bottom = True      # True= Show image text at bottom otherwise at top
image_font_size = 12          # Default= 12 Font text height in px for text on images
image_font_scale = 0.5        # Default= 0.5 Font scale factor that is multiplied by the font-specific base size.
image_font_thickness = 2      # Default= 2  Font text thickness in px for text on images
image_font_color = (255, 255, 255)  # Default= (255, 255, 255) White
image_bigger = 3.0            # Default= 3.0 min=0.1 Resize saved speed image by specified multiplier value
image_max_files = 0           # 0=off or specify MaxFiles to maintain then oldest are deleted  Default=0 (off)

image_sign_on = False
image_sign_show_camera = False
image_sign_resize = (1280, 720)
image_sign_text_xy = (100, 675)
image_sign_font_scale = 30.0
image_sign_font_thickness = 60
image_sign_font_color = (255, 255, 255)
image_sign_timeout = 5        # Keep the image sign for 5 seconds.

pageauc, 2023-02-14:
set config.py settings per below and give it a try.  Also make sure the
resolution for webcam is appropriate for your cam.   i suggest start at
320x240

WEBCAM = True         # Default= False False=PiCamera True= USB Webcam or
RTSP,IP Camera

# Web Camera Settings
WEBCAM_SRC = 0         # Default= 0   USB camera device connection number

angrytrkr, 2023-02-14:
Tried and still does not work. Do i need to do a full restart or just close the program. I have been using ./menubox.sh to run the program. I also do not see the power light on the camera staying on. I can use the camera with fswebcam just fine.

angrytrkr, 2023-02-14:
This is my usb list

pi@raspberrypi:~ $ lsusb
Bus 001 Device 005: ID 0c45:636b Microdia USB 2.0 Camera
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

pageauc, 2023-02-14:
make sure you have a zero not the letter o
Also you might try 1 and see what happens since I see you have a video1
showing on your original email

angrytrkr, 2023-02-14:
i am still getting nothing... Is there a way to tell if the program is actually using the camera? or you just have to try to trigger it by driving past it?

pageauc, 2023-02-14:
stop speed-cam.py.  Edit config.py and set

    align_cam_on = True
    align_delay_sec = 5

ensure config.py has the webcam and src set that you want to test.

If the webcam settings are working, this will take a calibration image
every 5 seconds to allow aligning the camera.
stop and restart speed-cam.py.  You should run speed-cam.py directly to see
verbose log information. see wiki
https://github.com/pageauc/speed-camera/wiki/How-to-Run

when testing and alignment is complete set

    align_cam_on = False

restart then retest and calibrate for real motion. see wiki for details.
https://github.com/pageauc/speed-camera/wiki

let me know results

angrytrkr, 2023-02-15:
I found when i manually run speed-cam.py i have library errors. I am having issues completing opencv3 install. When i start MAKE i usually fail at 17%. I am currently attempting a fresh install of cv3 to see if i can get past this issue. I will keep you updated. Thanks for staying in touch with me and helping me.

pageauc, 2023-02-15:
opencv should already be installed if you did a github bash install. You
could also install per

    sudo apt-get install -yq python3-opencv

Then try running

       cd ~/speed-camera
       python3 ./speed-cam.py

see if that works

angrytrkr, 2023-02-15:
This is what i get

speed-cam.py 11.26  written by Claude Pageau
Motion Track Largest Moving Object and Calculate Speed per Calibration.
----------------------------------------------------------------------
Loading  Wait ...
Traceback (most recent call last):
  File "/home/pi/speed-camera/./speed-cam.py", line 340, in <module>
    from picamera.array import PiRGBArray
  File "/usr/lib/python3/dist-packages/picamera/__init__.py", line 72, in <module>
    from picamera.exc import (
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 41, in <module>
    import picamera.mmal as mmal
  File "/usr/lib/python3/dist-packages/picamera/mmal.py", line 49, in <module>
    _lib = ct.CDLL('libmmal.so')
  File "/usr/lib/python3.9/ctypes/__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libmmal.so: cannot open shared object file: No such file or directory

pageauc, 2023-02-15:
Run

sudo raspi-config

under Interfaces options enable Legacy camera
reboot and try again.

if still issues run menubox.sh and run UPGRADE menu pick

angrytrkr, 2023-02-16:
Same error still.

pageauc, 2023-02-16:
it is possible you have a bad sd card raspi os image.  you may want to burn
fresh image on sd card or fesh card and try again.

angrytrkr, 2023-02-16:
I am using a libre board coult that be the issue? they provide their own image for the board so wondering if the image they provide is bad. AML-S905X-CC

angrytrkr, 2023-02-17:
Ended up reverting to Buster OS and everything is running fine now. Thanks for the help, too bad there isn't much available to get it working on newer.

closed 2023-02-17 by angrytrkr

pageauc, 2023-03-16:
FYI.  I purchase a libre potato and installed Bullseye 64.  I installed speed camera using curl. Ran with a Logitech web camera.  Had to make a minor code change to speed-cam.py due to not trapping an OSError.  Camera was setup as WEBCAM_SRC = 1 in config.py.  Also tried one of my security cameras using rtsp://LOGIN:PASSWORD@IP:PORT/PATH (per camera documentation) with no issues. this has been running fine for several weeks even after several large OS sudo apt update and upgrade.
Github has the latest code.  Run menubox.sh and do an UPGRADE
Regards Claude