pageauc/speed-camera #131: Difference between speed-cam.py and speed-cam-887.py?

johncblacker, 2023-04-07:
I see the subject files have different sizes...what's the difference between them and which one should be used?

closed 2023-04-07 by johncblacker

johncblacker, 2023-04-07:
I see, the 887 file is the 8.8.7 version while the speed-cam.py file is the latest version.  I'm running the latest version currently
but will be looking to convert this over to use the latest picamera2 library and the V3 camera which has greater resolution and autofocus as well.  The conversion is not that difficult.  In fact I intent to piggy back on the work that miguel grinberg has done in conjunction with using a Flask server.

pageauc, 2023-05-08:
I have updated speed-camera to support libcam.  Note config.py settings
have changed so after upgrade copy config.py.new to config.py per upgrade
instructions.  I have tested with rpi 64 bit bullseye and also libre potato
(who thinks up these names.  I am looking at adding reading from video file
and also adding support for vehicle recognition.
Regards Claude ....