pageauc/speed-camera #133: Inconsistent Speeds

kummerr, 2023-06-17:
Hi Claude,

Thanks for sharing this! I think I have this calibrated correctly, but I'm getting inconsistent speeds. For example it records a speed of 18.4 mph and less than a second later it records the same car at 142.5 mph.  I probably have something misconfigured somewhere. Would you mind taking a look at my speed cam settings and let me know if you can see what I am doing wrong?

Thanks,

Bob

# ---------------- User Configuration Settings for speed-cam.py ---------------------------------
#         Ver 12.00 speed-cam.py Variable Configuration Settings

#######################################
#  speed-cam.py Variable Settings
#  Default is 320x240 image stream size
#  Note motion track crop area is now
#  auto configured so you may not need to use a plugin
#  to change resolution.
#######################################

# Calibration Settings
# --------------------
calibrate = False         # Create a calibration image file with calibration hash markers 10 px per mark
align_cam_on = False     # Default=False  True Saves alignment image to help with camera pointing
align_delay_sec = 5      # Default=5 seconds delay between each alignment image

cal_obj_px_L2R = 270      # L2R Moving Objects, Length of a calibration object in pixels
cal_obj_mm_L2R = 3850.0    # L2R Moving Objects, Length of the calibration object in millimetres

cal_obj_px_R2L = 260      # R2L Moving Objects, Length of a calibration object in pixels
cal_obj_mm_R2L = 3850.0    # R2L Moving Objects, Length of the calibration object in millimetres

# Camera Settings
# ---------------
CAMERA = "ipcam"      # valid values are usbcam, ipcam, pilibcam, pilegcam
CAM_LOCATION = 'Driveway'    # Specify an address, physical location Etc for camera

# Libcam (Bullseye or later) and Legacy Pi Camera Settings
# --------------------------------------------------------
CAMERA_WIDTH = 320     # Image stream width for opencv motion scanning Default=320
CAMERA_HEIGHT = 240    # Image stream height for opencv motion scanning  Default=240
CAMERA_FRAMERATE = 15  # Default= 20 Frame rate for video stream V2 picam can be higher
CAMERA_ROTATION = 0    # Rotate camera image valid values are 0, 90, 180, 270
CAMERA_VFLIP = True    # Flip the camera image vertically if required
CAMERA_HFLIP = True    # Flip the camera image horizontally if required

# USB, WEB and  RTSP IP Cam Settings
# ---------------------------------
USBCAM_SRC = 1         # Used if CAMERA = "usbcam"
IPCAM_SRC = "rtsp://admin:password@192.168.86.129:554/cam/realmonitor?channel=1&subtype=2"   # Set low res stream per IP Cam Docs and config
WEBCAM_WIDTH = 320     # Default= 320 USB, RTSP cam image stream width
WEBCAM_HEIGHT = 240    # Default= 240 USB, RTSP cam image stream height
WEBCAM_HFLIP = False   # Default= False USB Webcam flip image horizontally
WEBCAM_VFLIP = False   # Default= False USB Webcam flip image vertically
                       # IMPORTANT Webcam Streaming Performance Hit if Stream Flipped.

# Note if tested speed is too low increase appropriate cal_obj_mm  value and redo speed test for desired direction.
# IMPORTANT - If plugins Enabled Edit Settings in specified plugin file located in plugins folder.

# Plugins overlay the config.py variable settings
# -----------------------------------------------
pluginEnable = True
pluginName = "webcam720" # Specify filename in plugins subfolder without .py extension per below
                        # picam240, webcam240 (Recommended for RPI2 or greater)
                        # picam480, webcam480, picam720, webcam720  (can use RPI3 but Test)
                        # picam1080   (Experimental Not Recommended)
                        # secpicam480, secwebcam480 (Experimental no CSV entries)

# Display opencv windows on gui desktop
# gui_window_on suppresses All Windows if False
# ----------------------------------------------
[... 117 lines cut ...]
# Web Server settings
# -------------------
web_server_port = 8080        # Default= 8080 Web server access port eg http://192.168.1.100:8080
web_server_root = "media"     # Default= "media" webserver root path to webserver image/video sub-folders
web_page_title = "SPEED-CAMERA Media"  # web page title that browser show (not displayed on web page)
web_page_refresh_on = True    # False=Off (never)  Refresh True=On (per seconds below)
web_page_refresh_sec = "900"  # Default= "900" seconds to wait for web page refresh  seconds (15 minutes)
web_page_blank = False        # True Starts left image with a blank page until a right menu item is selected
                              # False displays second list[1] item since first may be in progress

# Left iFrame Image Settings
# --------------------------
web_image_height = "768"       # Default= "768" px height of images to display in iframe
web_iframe_width_usage = "70%" # Left Pane - Sets % of total screen width allowed for iframe. Rest for right list
web_iframe_width = "80%"      # Desired frame width to display images. can be eg percent "80%" or px "1280"
web_iframe_height = "80%"     # Desired frame height to display images. Scroll bars if image larger (percent or px)

# Right Side Files List
# ---------------------
web_max_list_entries = 0           # 0 = All or Specify Max right side file entries to show (must be > 1)
web_list_height = web_image_height # Right List - side menu height in px (link selection)
web_list_by_datetime = True        # True=datetime False=filename
web_list_sort_descending = True    # reverse sort order (filename or datetime per web_list_by_datetime setting

# ---------------------------------------------- End of User Variables -----------------------------------------------------

carterw, 2023-08-01:
I'm not seeing that here. What kind of camera are you using and what Raspberry Pi?

kummerr, 2023-08-01:
For the camera I'm using the 640x480 substream of a Dahua IPC-HDW5231R-ZE.  The application is running on a VirtualBox RPi VM.

carterw, 2023-08-01:
That's a pretty exotic setup. Are you sure you are getting image frames at a constant and accurate rate? The application depends on a steady stream of images that all come in on time so it can measure pixels over time.

kummerr, 2023-08-02:
I thought so, but now that you bring this up I'm not so sure. I know that the camera stream is working well in Blue Iiris.

carterw, 2023-08-02:
If I were you I would write a little diagnostic code that monitors the amount of time between the incoming images and looks for intervals that are out of the ordinary. Maybe at the top of the "while still_scanning" loop.

closed 2023-11-06 by pageauc