pageauc/speed-camera #134: double frame read

carterw, 2023-08-01:
It looks like two image frames are being acquired for every cycle through the scanning loop and the first one is not used.

On line 1326 there is the scanning loop;
`while still_scanning:  # process camera thread images and calculate speed` 

Then there's a frame read;
`image2 = vs.read()  # Read image data from video steam thread instance`
And then speed_get_contours() is called, passing in the frame that was read;
 `grayimage1, contours = speed_get_contours(image2, grayimage1)`

But then speed_get_contours() ignores the frame that was passed in and does another read in the `while not image_ok:` loop.

Was this intentional? It seems like that second frame read could be eliminated.

pageauc, 2023-08-02:
function
def speed_get_contours(image, grayimage1):

Does the contour processing.  It receives the current frame image and
converts it to grayscale and compares it to previous grayscale array to
calculate contours.
When finished the previous grayscale array is updated to the most recent
grayscale line 1218, that is returned by the function ready for the next
loop.
This is done to reduce processing and is by design.

Anyone can send changes via github push for review if they have an improved
method.

Claude

carterw, 2023-08-02:
Yes I understand what the method does. I was pointing out that speed_get_contours() ignores the incoming frame and performs a second frame read in a loop before a crop is attempted. Maybe the vs.read() could be moved to the bottom of that loop? But if you are happy with it as-is that's fine with me.

pageauc, 2023-08-05:
Thanks for your correct explanation. 
See speed-cam.py ver 12.01 for updated logic 
Claude ...

carterw, 2023-08-05:
Thanks for the helpful fix!

closed 2023-08-05 by carterw

pageauc, 2023-08-05:
Note 12.02 replaces np-std with np.median for calculating ave speed.  

Speed camera is written as an object tracker.  I will look at adding an option to use vehicle recognition (opencv haarcascade) that can improve speed calculation.

Claude.