pageauc/speed-camera #136: How to set focus for camera on module v3

yaomtc, 2023-08-21:
The libcamera-apps have this option:

	 --lens-position				Set the lens to a given position <string>

> Moves the lens to a fixed focal distance, normally given in dioptres (units of 1 / distance in metres). We have
> * 0.0 will move the lens to the "infinity" position
> * Any other number: move the lens to the 1 / number position, so the value 2 would focus at approximately 0.5m
> * default - move the lens to a default position which corresponds to the hyperfocal position of the lens.

Is it possible to use this capability with speed-camera?

pageauc, 2023-08-21:
In the file  strmpilibcam.py  you can add your setting in the appropriate
location in the def __init__ function
I do not have a v3 camera.  If you get things working then send me details
and I could add as a variable that can be passed so it can be easily set in
config.py
Claude ...

pageauc, 2023-08-21:
Not sure if manual or auto focus would work best.  Try

    picam2.set_controls({"AfMode": controls.AfModeEnum.Manual,
"LensPosition": 0.0})

or Auto

     picam2.set_controls({"AfMode": controls.AfModeEnum.Continuous})

example in strmplibcam.py  *(cannot test)*

        self.picam2.configure(self.picam2.create_preview_configuration(
                              main={"format": 'XRGB8888',
                              "size": size},
                              raw={"size":self.picam2.sensor_resolution},
                             "AfMode"=controls.AfModeEnum.Continuous,
                              transform=Transform(vflip=vflip,
                                                  hflip=hflip)))

let me know results.  Also try manual.

pageauc, 2023-08-21:
You could also just try adding the focus code as a separate line per first
email below main self.picam2

eg
        self.picam2.configure(self.picam2.create_preview_configuration(
                              main={"format": 'XRGB8888',
                              "size": size},
                              raw={"size":self.picam2.sensor_resolution},
                              transform=Transform(vflip=vflip,
                                                  hflip=hflip)))
        self.picam2.set_controls({"AfMode":
controls.AfModeEnum.Continuous})  # Set Auto focus
        self.picam2.start()

yaomtc, 2023-08-29:
>     "AfMode"=controls.AfModeEnum.Manual,"LensPosition": 0.0,
>     ^
>     SyntaxError: expression cannot contain assignment, perhaps you meant "=="?

Then when I try the focus code as a separate line:

>         self.picam2.set_controls({"AfMode":controls.AfModeEnum.Continuous}) # Set auto focus
>     NameError: name 'controls' is not defined

pageauc, 2023-08-29:
I don't have a camera with focus.  You can try referencing the picamer2
libcamera documentation.  This should be a good source of information.
https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf
Also take a look at the github examples
https://github.com/raspberrypi/picamera2/tree/main/examples

Modify speed camera local copy of strmpilibcam.py  (you may want to save a
backup copy of original)

https://github.com/pageauc/speed-camera/blob/master/strmpilibcam.py

Insert test camera code as a separate line in the def __init__  just before
the self.picam2.start() line

Also be aware that picamera2 python lib is still in Beta

I would appreciate an update on your testing.

Best Wishes
Claude ...

pageauc, 2023-08-29:
Take a look a picamera2 manual page 29 in focus section 5.2
Examples show focus code after camera start so you might try that also make
sure space are ok since that sometimes makes a difference.

closed 2023-11-06 by pageauc

UnderpantsGnome, 2024-04-23:
Hi @pageauc Just in case this comes up again, [this article](https://www.tomshardware.com/how-to/raspberry-pi-camera-module-3-python-picamera-2) has some good info in it, and this is all I needed to change to get it to work.

```diff
diff --git a/strmpilibcam.py b/strmpilibcam.py
index 7765a44..d8f95b2 100644
--- a/strmpilibcam.py
+++ b/strmpilibcam.py
@@ -2,6 +2,7 @@
 # Import required libraries
 from picamera2 import Picamera2
 from libcamera import Transform
+from libcamera import controls
 import time
 from threading import Thread
 
@@ -38,6 +39,10 @@ class CamStream:
                               raw={"size":self.picam2.sensor_resolution},
                               transform=Transform(vflip=self.vflip,
                                                   hflip=self.hflip)))
+
+        #self.picam2.set_controls({"AfMode": controls.AfModeEnum.Continuous, "AfSpeed": controls.AfSpeedEnum.Fast})
+        self.picam2.set_controls({"AfMode": controls.AfModeEnum.Manual,"LensPosition": 12.0})
+
         self.picam2.start()
         time.sleep(2) # Allow camera time to warm up
```