pageauc/speed-camera #137: Latest update speed detection issue?

grunta56, 2023-09-07:
After running the update released on about 2 Sept or so I am only detecting very slow objects such as pedestrians and animals. No traffic. Was working perfectly prior to this. I've spent some time on this including completely fresh install and going over my config but without success. Just wondering if anyone else has similar issue? Thankyou. Grant

pageauc, 2023-09-07:
> YouTube Channel at https://www.youtube.com/user/pageaucp
> GitHub Repository at https://github.com/pageauc

I run on low end RPI's (Headless) and have not noticed any change.  Update
was just to improve/fix GUI window displays.  Will check to see if I missed
anything.

pageauc, 2023-09-07:
Released ver 12.05 now only makes a copy of image2 when required. Making an
extra copy is required to fix GUI windows, align and calibrate features.
see line 1328 of speed-cam.py

        if gui_window_on or align_cam_on or calibrate:
            image2_copy = image2  # make a copy of current image2 when
needed

Give this a try and let me know.
Claude ....

grunta56, 2023-09-07:
Still no luck after installing this Claude.

pageauc, 2023-09-09:
What was your previous version of speed-cam.py that worked OK and also let me know operating system version 

    uname -a

grunta56, 2023-09-12:
apologies for the delayed response Claude. I am away from the installation for a number of days but will get back to this as soon as I return. From memory the version of speed-cam.py was the first release that ran on Bullseye with the Raspi HQ Camera. If in the meantime no-one else reports the issue then I guess its at my end and perhaps I would be best advised to do a fresh Bullseye install on a fresh formatted card and start over.