pageauc/speed-camera #138: Not saving vehicles

victor7376, 2023-09-12:
Hi,

I love this program but I've got an issue with it lol.

The server & tracking starts up with no problem & gets the feed from my rtsp camera (Tapo C200 - 640x360) & the log shows it is seeing the vehicles going past & calculating the speeds, but its not saving them.  But as soon as someone walks past its detecting them & saving those.  I've tried to adjust the settings to only save those that are going over a certain speed (I have it showing MPH) but max_speed_over I wasnt sure if this was in MPH or KPH so put it to 10.

Config:
`# ---------------- User Configuration Settings for speed-cam.py ---------------------------------
#         Ver 12.00 speed-cam.py Variable Configuration Settings

#######################################
#  speed-cam.py Variable Settings
#  Default is 320x240 image stream size
#  Note motion track crop area is now
#  auto configured so you may not need to use a plugin
#  to change resolution.
#######################################

# Calibration Settings
# --------------------
calibrate = False        # Create a calibration image file with calibration hash markers 10 px per mark
align_cam_on = False     # Default=False True Saves align_cam.jpg See Web page recent align cam
align_delay_sec = 2      # Default=2 seconds delay between each alignment image (disables motion tracking)

cal_obj_px_L2R = 180     # L2R Moving Objects, Length of a calibration object in pixels
cal_obj_mm_L2R = 4000    # L2R Moving Objects, Length of the calibration object in millimetres

cal_obj_px_R2L = 180     # R2L Moving Objects, Length of a calibration object in pixels
cal_obj_mm_R2L = 4000    # R2L Moving Objects, Length of the calibration object in millimetres

# Camera Settings
# ---------------
CAMERA = "ipcam"      # valid values are usbcam, ipcam, pilibcam, pilegcam
CAM_LOCATION = 'None'    # Specify an address, physical location Etc for camera

# Libcam (Bullseye or later) and Legacy Pi Camera Settings
# --------------------------------------------------------
CAMERA_WIDTH = 320     # Image stream width for opencv motion scanning Default=320
CAMERA_HEIGHT = 240    # Image stream height for opencv motion scanning  Default=240
CAMERA_FRAMERATE = 22  # Default= 20 Frame rate for video stream V2 picam can be higher
CAMERA_ROTATION = 0    # Rotate camera image valid values are 0, 90, 180, 270
CAMERA_VFLIP = True    # Flip the camera image vertically if required
CAMERA_HFLIP = True    # Flip the camera image horizontally if required

# USB, WEB and RTSP IP Cam Settings
# ---------------------------------
USBCAM_SRC = 1         # Used if CAMERA = "usbcam"
IPCAM_SRC = "rtsp://RoydzNAS:pussycat76777578@192.168.68.144:554/stream2"  # Set low res stream per IP Cam Docs and config
WEBCAM_WIDTH = 640     # Default= 320 USB, RTSP cam image stream width
WEBCAM_HEIGHT = 360    # Default= 240 USB, RTSP cam image stream height
WEBCAM_HFLIP = False   # Default= False USB Webcam flip image horizontally
WEBCAM_VFLIP = False   # Default= False USB Webcam flip image vertically
                       # IMPORTANT Webcam Streaming Performance Hit if Stream Flipped.

# Note if tested speed is too low increase appropriate cal_obj_mm  value and redo speed test for desired direction.
# IMPORTANT - If plugins Enabled Edit Settings in specified plugin file located in plugins folder.

# Plugins overlay the config.py variable settings
# -----------------------------------------------
pluginEnable = True
pluginName = "picam240" # Specify filename in plugins subfolder without .py extension per below
                        # picam240, webcam240 (Recommended for RPI2 or greater)
                        # picam480, webcam480, picam720, webcam720  (can use RPI3 but Test)
                        # picam1080   (Experimental Not Recommended)
                        # secpicam480, secwebcam480 (Experimental no CSV entries)

# Display opencv windows on gui desktop
# gui_window_on suppresses All Windows if False
# ----------------------------------------------
gui_window_on = True # True= Turn On All desktop GUI openCV windows. False=Don't Show (req'd for SSH) .
[... 118 lines cut ...]
# -------------------
web_server_port = 8080        # Default= 8080 Web server access port eg http://192.168.1.100:8080
web_server_root = "media"     # Default= "media" webserver root path to webserver image/video sub-folders
web_page_title = "SPEED-CAMERA Media"  # web page title that browser show (not displayed on web page)
web_page_refresh_on = True    # False=Off (never)  Refresh True=On (per seconds below)
web_page_refresh_sec = "400"  # Default= "900" seconds to wait for web page refresh  seconds (15 minutes)
web_page_blank = False        # True Starts left image with a blank page until a right menu item is selected
                              # False displays second list[1] item since first may be in progress

# Left iFrame Image Settings
# --------------------------
web_image_height = "768"       # Default= "768" px height of images to display in iframe
web_iframe_width_usage = "70%" # Left Pane - Sets % of total screen width allowed for iframe. Rest for right list
web_iframe_width = "100%"      # Desired frame width to display images. can be eg percent "80%" or px "1280"
web_iframe_height = "100%"     # Desired frame height to display images. Scroll bars if image larger (percent or px)

# Right Side Files List
# ---------------------
web_max_list_entries = 0           # 0 = All or Specify Max right side file entries to show (must be > 1)
web_list_height = web_image_height # Right List - side menu height in px (link selection)
web_list_by_datetime = True        # True=datetime False=filename
web_list_sort_descending = True    # reverse sort order (filename or datetime per web_list_by_datetime setting

# ---------------------------------------------- End of User Variables -----------------------------------------------------
`

pageauc, 2023-09-12:
> IPCAM_SRC = ***@***.***:554/stream2"

Everything looks OK.  Not sure if your IP cam stream is low resolution.
Having a large image really slows down opencv.

Also speed-cam.py is at ver 12.05 so upgrade if lower.
I have run speed-cam.py on RPI 3 quad core with IP Camera stream with no
issues.  Also have a single core RPI using RPI Legacy cam an other than
missing the odd vehicle it works fine.  My IP cams have a low resolution
640x480 stream and that is what I use.
Please check you IP Cam Docs and pick the lowest res stream size.  Even
720p can cause slow RPI.
Claude ...

victor7376, 2023-09-12:
I've tweaked the settings and it seems to have now worked.

The Tapo C200's lowest resolution is 640x360 as its a 1080p camera.

closed 2023-09-12 by pageauc

victor7376, 2023-09-14:
Having to reopen my issue, as after a few hours of working the app crashes after not getting any response for 60 seconds.  I've confirmed the camera is working on not only the TPLink app, but also the cctv software I use (AgentDVR).  I even tried using the feed from agentdvr into the speed camera, but the speed camera crashed after a few hours of use with the same error.

Today I tried to use a usb webcam (c270) but when I run the software the webcam blinks on then off.

Not sure what is happening.  I did upgrade the software.  I've even tried on Ubuntu to see if that made any difference, but its the same issue.

victor7376, 2023-09-14:
The error message is:

2023-09-13 20:15:12 ERROR    speed_get_contours image Stream Image is Not Complete. Cannot Crop. Retry.

after 60 times then crashes