pageauc/speed-camera #139: image_jpeg_optimize = True

carterw, 2023-09-13:
I was looking at the image storage code and noticed that the cv2.imwrite() section around line 1576 in speed-cam.py is always throwing an exception. It is caught and the image write always happens with no optimizations. The code looks like this;

` cv2.imwrite(
                                            filename,
                                            big_image,
                                            [
                                                int(cv2.IMWRITE_JPEG_QUALITY),
                                                image_jpeg_quality,
                                                int(cv2.IMWRITE_JPEG_OPTIMIZE),
                                                image_jpeg_optimize,
                                            ]`

The problem appears to be that cv2.IMWRITE_JPEG_OPTIMIZE wants to see a 0 or 1 argument, and what is being passed in through _image_jpeg_optimize_ is the value _True_. Setting it to 1 results in success.

pageauc, 2023-09-13:
Pushed ver 12.06 that looks like it fixed issue.  Honestly I could not see image improvement.  My default config is normally set to Not optimized and run machines headless and speed-cam.py in background. Originally put image_jpeg_optimize as imwrite parameter since True and 1 are normally the same.  I did not see exception during original testing quit a few years ago so maybe openCV changed behavior,  Anyway you are the first to point this out. 
Thanks for your issue.  Let me know if new version works OK for you.
Regards Claude.

carterw, 2023-09-14:
I don't know if image_jpeg_optimize helps either, I just happened to be stepping through that section of code and noticed the exception.

closed 2023-09-14 by carterw