pageauc/speed-camera #140: Updated has errors

victor7376, 2023-09-18:
Hey m8,

I’ve just updated to the latest version, the new one isn’t downloading the strmcam.py module so had to download that manually.

plus the speed-cam.py has two errors from line 414 with too many “ in it.

pageauc, 2023-09-18:
A few implementation issues.  strmcam.py was not getting installed on
upgrade.  Also fixed comments on make_media_dirs()
Github repo has been updated.  Sorry for the problems.  Did not expect
anyone to access so quickly.
Let me know if you find any other issues.
Thanks Claude ...

victor7376, 2023-09-18:
As soon as I fixed the errors manually, it’s all working.  Cheers.

I just happened to see there was an update so downloaded it just now lol.

pageauc, 2023-09-19:
I have tested end to end with several RPI's.  Worked fine but let me know
if you find any bugs.  This was a pretty big change renaming constants and
some functions.  Also the strmcam changes.  I still think strmpilibcam can
be faster.
Thanks
Regards Claude

victor7376, 2023-09-19:
No problem, will do.  I keep changing between the rtsp camera feed & the picamera I got from amazon due to when the sun shines on the window the program doesn’t always save the detections even though the log is showing the detection of the vehicles & the speeds.
 Only real issue I had yesterday was a white line within the text on the images.  But will be doing more testing tomorrow.

victor7376, 2023-09-19:
Found another bug lol

when you start the webserver, it says the IP but then it’s missing the port number at the end.

kummerr, 2023-09-19:
After updating to the latest version the web server won’t even start.

victor7376, 2023-09-19:
I have downloaded on another raspberry this morning and it works just fine except for the port number bug I found.

kummerr, 2023-09-19:
So I backed up my config.py file and deleted everything and reinstalled the application using the curl command and was able to get the web server to start. However, when I go into settings to reconfigure everything the camera section is not there. Did it get moved or am I missing something?

victor7376, 2023-09-19:
It’s got moved to configcam.py

kummerr, 2023-09-19:
Got it, thank you!

pageauc, 2023-09-19:
Just ran webserver.py (latest github ver) I do not see any problems. see webserver.py output below.  Running on bullseye rpi4 with latest update and upgrade.

Importing Configuration Variables from File /home/pi/speed-camera/config.py
----------------------------------------------------------------
webserver.py ver 13.00 written by Claude Pageau modified by Alexandre Strube for python3 compatibility
---------------------------- Settings --------------------------
Server  - WEB_PAGE_TITLE   = SPEED-CAMERA Media
          WEB_SERVER_ROOT  = /home/pi/speed-camera/media
          WEB_SERVER_PORT  = 8080
Content - WEB_IMAGE_HEIGHT = 768 px (height of content)
          WEB_IFRAME_WIDTH = 100%  WEB_IFRAME_HEIGHT = 100%
          WEB_IFRAME_WIDTH_PERCENT = 70% (of avail screen)
          WEB_PAGE_REFRESH_SEC = 900  (default=180 sec)
          WEB_PAGE_BLANK_ON = False ( True=blank left pane until item selected)
Listing - WEB_MAX_LIST_ENTRIES = 0 ( 0=all )
          WEB_LIST_BY_DATETIME_ON = True  sort_decending = True
----------------------------------------------------------------
From a computer on the same LAN. Use a Web Browser to access this server at
Type the URL below into the browser url bar then hit enter key.

                 http://192.168.1.120:8080
                 http://192.168.1.141:8080

IMPORTANT: If You Get - socket.error: [Errno 98] Address already in use
           Check for Another app using port or Wait a minute for webserver to timeout and Retry.
              ctrl-c to exit this webserver script
----------------------------------------------------------------

This ver running on rpi 3 bullseye

Importing Configuration Variables from File /home/pi/speed-camera/config.py
----------------------------------------------------------------
webserver.py ver 13.00 written by Claude Pageau modified by Alexandre Strube for python3 compatibility
---------------------------- Settings --------------------------
Server  - WEB_PAGE_TITLE   = SPEED-CAMERA Media
          WEB_SERVER_ROOT  = /home/pi/speed-camera/media
          WEB_SERVER_PORT  = 8080
Content - WEB_IMAGE_HEIGHT = 768 px (height of content)
          WEB_IFRAME_WIDTH = 100%  WEB_IFRAME_HEIGHT = 100%
          WEB_IFRAME_WIDTH_PERCENT = 70% (of avail screen)
          WEB_PAGE_REFRESH_SEC = 900  (default=180 sec)
          WEB_PAGE_BLANK_ON = False ( True=blank left pane until item selected)
Listing - WEB_MAX_LIST_ENTRIES = 0 ( 0=all )
          WEB_LIST_BY_DATETIME_ON = True  sort_decending = True
----------------------------------------------------------------
From a computer on the same LAN. Use a Web Browser to access this server at
Type the URL below into the browser url bar then hit enter key.

                 http://192.168.1.164:8080

IMPORTANT: If You Get - socket.error: [Errno 98] Address already in use
           Check for Another app using port or Wait a minute for webserver to timeout and Retry.
              ctrl-c to exit this webserver script
----------------------------------------------------------------

Let me know what ver rpi and OS ver.

pageauc, 2023-09-19:
I have added comment in the config.py regarding camera settings now being stored in configcam.py

victor7376, 2023-09-19:
It’s when you launch the webserver from the menu box that there’s no port number.

pageauc, 2023-09-19:
OK that makes sense.  Will update menubox.sh.

pageauc, 2023-09-19:
menubox.sh web port display fixed.  New update on github.  Tested and works now.

closed 2023-09-19 by pageauc