pageauc/speed-camera #142: Graph program

victor7376, 2023-09-21:
Hi,  Just noticed that the graph report scripts dont work if.  Think its because you removed the SPEED_MPH from the config.py

victor7376, 2023-09-21:
Fixed the issue in my branch ;)

Changed SPEED_MPH to MO_SPEED_MPH_ON  now the graph script works

pageauc, 2023-09-21:
Fixed both reports and updated Repo.  Currently I am trying to get plugins to work.  May have to rethink. 
Thanks for your help

victor7376, 2023-09-22:
what is the issue with the plugin as when I have them enabled it says it loads the one I ask for.

pageauc, 2023-09-22:
I have redesigned plugins and eliminated the need for configcam.py.  All settings will be in config.py.  Less complicated following KISS principle.  Updated a few plugins for testing but remainder need to be updated for new variable names.  Should be done in a day or so.  Also fine tuning messages.
Thanks for you patience and help.
Regards  Claude

pageauc, 2023-09-22:
I have uploaded ver 13.05.  Includes bug fixes, working plugins, Etc.  All settings are again stored in config.py.  You may notice a few new settings.  I recommend a clean install otherwise To get new plugins delete old ones.  Also recommend deleting configcam.py since it is no longer used.   Raise an issue if you encounter a bug or problem.
Thanks for your feedback. 
Cl;aude ....

closed 2023-09-28 by pageauc