pageauc/speed-camera #143: RTSP camera

victor7376, 2023-09-27:
Hi m8,

For the past week I've been testing the setup with a raspberry camera & its been working perfectly with the calibration & also detecting all the speeds etc.

So today I've tried once again to see if I could get my IP camera to work.  No matter what I do, it just doesnt save the images or detect the speeds on around 80 - 90% of the vehicles going past.

When a vehcile goes past I can see the read out via the terminal using the 'tail -f' command on the log for example:

2023-09-27 20:47:19 INFO     speed_camera  Out - 0/6 xy(451,8) Max D=139>=24px C=2 118x85=4065 sqpx L2R
2023-09-27 20:47:19 INFO     speed_camera New  - 0/6 xy(371,6) Start New Track
2023-09-27 20:47:19 INFO     speed_camera  Out - 0/6 xy(400,5) Max D=29>=24px C=1 190x81=5687 sqpx L2R
2023-09-27 20:47:19 INFO     speed_camera New  - 0/6 xy(428,5) Start New Track
2023-09-27 20:47:19 INFO     speed_camera  Out - 0/6 xy(453,4) Max D=25>=24px C=1 137x90=6124 sqpx L2R
2023-09-27 20:47:19 INFO     speed_camera New  - 0/6 xy(504,5) Start New Track
2023-09-27 20:47:19 INFO     speed_camera  Add - 1/6 xy(527,7) 18.74 mph D=23/24 C=1 63x81=3635 sqpx L2R
2023-09-27 20:47:19 INFO     speed_camera  Add - 2/6 xy(547,6) 17.33 mph D=20/24 C=1 43x76=2201 sqpx L2R
2023-09-27 20:47:19 INFO     speed_camera  Add - 3/6 xy(566,5) 15.91 mph D=19/24 C=1 24x67=920 sqpx L2R

The camera is a Tapo C200 using RTSP with a quality of 640x360 (which is the lowest it'll go)

pageauc, 2023-09-27:
It should work since I tested on a topodome rtsp camera.  You can try
reducing the size of the crop area.
speed-cam.py auto detects RTSP image shape approx line 1854 using opencv
since config.py cannot control the RTSP image size.
You can however control the crop area.  Try to make this as small as
possible. Set config.py ALIGN_CAM_ON = True.  You can view recent on
webserver.  Then set MO_CROP_AUTO_ON = False then adjust x-left, right,
y-upper and lower.  You might want to do this in a plugin similar to a
480 plugin. but for an ip cam. start with settings below as staring point.

    MO_CROP_X_LEFT = 100          # Default=50 comment variable for auto
calculate
    MO_CROP_X_RIGHT = 540         # Default=250 comment variable for auto
calculate
    MO_CROP_Y_UPPER = 170         # Default=90 comment variable for auto
calculate
    MO_CROP_Y_LOWER = 310         # Default=150 comment variable for auto
calculate

Claude

pageauc, 2023-09-27:
Another possible factor is wifi speed.  Distance can reduce throughput.  I tested my RTSP cam with a wired connection.

victor7376, 2023-09-27:
Unfortunately this camera is WiFi only, but I have excellent speeds.

But I’ll do some more testing with two pi 4’s tomorrow.  Think I’ll try and get the cropped area as small as the pi camera does.  Once I’ve done testing I’ll post back.

Out of interest which camera do you use?

pageauc, 2023-09-28:
I use topodome cameras.

pageauc, 2023-09-28:
FYI Here is a sample of my speed plugin.  Performance is good on wifi only.  Note I still have camera info on.

[speed480.py.txt](https://github.com/pageauc/speed-camera/files/12749728/speed480.py.txt)
[image]

pageauc, 2023-09-28:
Here is copy of speed480.py plugin FYI

```
# ---------------- User Configuration Settings for speed-cam.py ---------------------------------
#         Ver 13.05 speed-cam.py picam480 Stream Variable Configuration Settings

#######################################
#    speed-cam.py plugin settings
#######################################

CALIBRATE_ON = False         # Create a calibration image file with calibration hash markers 10 px per mark
ALIGN_CAM_ON = False         # Default=False  True Saves alignment image to help with camera pointing
CAMERA = "rtspcam"          # valid values usbcam, rtspcam, pilibcam, pilegcam

RTSPCAM_SRC = "rtsp://admin:my_pass@192.168.1.210:554/12"  # Set per IP Cam Docs and config see example below
                                                  # rtsp://admin:myped@192.168.1.100:554/12

# Camera Image Stream Settings
IM_SIZE = (640, 352)   # Image resolution width, height pixels

# Motion Tracking Window Crop Area Settings
# -----------------------------------------
# To see motion tracking crop area on images, Set variable IM_SHOW_CROP_AREA_ON = True
# Set ALIGN_CAM_ON = True to Help with adjusting settings.
MO_CROP_AUTO_ON = False       # True enables rough Auto Calculation of Motion Crop Area
MO_CROP_X_LEFT = 170          # Default=50 comment variable for auto calculate
MO_CROP_X_RIGHT = 470         # Default=250 comment variable for auto calculate
MO_CROP_Y_UPPER = 130         # Default=90 comment variable for auto calculate
MO_CROP_Y_LOWER = 230         # Default=150 comment variable for auto calculate

# Motion Event Settings
# ---------------------
MO_TRACK_EVENT_COUNT = 6    # Default= 6 Number of Consecutive Motion Events to trigger speed photo. Adjust to suit.
                            # Suggest single core cpu=4-7 quad core=8-15 but adjust to smooth erratic readings due to contour jumps
MO_MIN_AREA_PX = 100        # Default= 200 Exclude all contours less than or equal to this sq-px Area
MO_LOG_OUT_RANGE_ON = True  # Default= True Show Out of Range Events per x_diff settings below False= Off
MO_MAX_X_DIFF_PX = 30       # Default= 20 Exclude if max px away >= last motion event x position
MO_MIN_X_DIFF_PX = 1        # Default= 1 Exclude if min px away <= last event x position
MO_X_LR_SIDE_BUFF_PX = 10   # Default= 10 Divides motion Rect x for L&R Buffer Space to Ensure contours are in
MO_TRACK_TIMEOUT_SEC = 0.5  # Default= 0.5 Optional seconds to wait after track End (Avoids dual tracking)
MO_EVENT_TIMEOUT_SEC = 0.3  # Default= 0.3 seconds to wait for next motion event before starting new track
MO_MAX_SPEED_OVER = 0       # Exclude track if Speed less than or equal to value specified 0=All
                            # Can be useful to exclude pedestrians and/or bikes, Etc or track only fast objects

# Camera Image Settings
# ---------------------
IM_FRAMERATE = 20
IM_SHOW_CROP_AREA_ON = True      # True= Display motion detection rectangle area on saved images
IM_SHOW_SPEED_FILENAME_ON = True # True= Include speed value in filename
IM_SHOW_TEXT_ON = True           # True= Show Text on speed images   False= No Text on images
IM_SHOW_TEXT_BOTTOM_ON = True    # True= Show image text at bottom otherwise at top
IM_FONT_SIZE_PX = 16             # Default= 12 Font text height in px for text on images
IM_FONT_SCALE = 0.5              # Default= 0.5 Font scale factor that is multiplied by the font-specific base size.
IM_FONT_THICKNESS = 2            # Default= 2  Font text thickness in px for text on images
IM_FONT_COLOR = (255, 255, 255)  # Default= (255, 255, 255) White
IM_BIGGER = 1.8                  # Default= 3.0 min=0.1 Resize saved speed image by specified multiplier value

# ---------------------------------------- End of User Variables ------------------------------------------

```

victor7376, 2023-09-28:
Thanks for the info, think that’s where my camera is different as my property is virtually on top of the road.

Example:
[image]

pageauc, 2023-09-28:
Yes too close but looks like you are getting successful tracks.  You may
want to tilt the camera view horizontal to the road.  Then set a crop area
to cover the minimum area to get good tracks.  opencv  will probably track
a part of the vehicle like hub cap or something else.  If you tune the
config you may still be OK.

victor7376, 2023-09-28:
This is what I dont understand, the Pi Camera has a small window for tracking with a resolution of 320x240 & works perfectly reading virtually 80 - 90% of vehicles, but this camera (Tapo C200) which has a resolution of 640x360 & a wider window to track doesnt get many.

They're both in the same position

Example of the Pi Camera:

[image]

pageauc, 2023-09-28:
Has to track consistently moving contours.  If too close contours can move around.  Also multiple objects may cause tracking issues.

konzty, 2023-10-19:
Hello Victor,

judging from your example pictures your pi camera uses a standard angle lens, objects move relatively linear through the picture (linear speed in pixels per second). On a wide angle lens, especialy fish-eye lenses, objects in the middle appear big, on the picture edges the same objects appear much much smaller. An object travelling 100 kmh (27.7 m/s) on the road might travel 10 pixel/s on the edge but getting closer to the middle of the picture the object grows bigger (in pixels) - also distances grow bigger ... so it might travel 50 pixels/s in the middle of the picture...

I do use a fisheye lense succesfully relatively close to the road and I had to use the following settings in my config.py:

I used my own car for calibration, the fisheye and it's strong non-linear geometry causes a huge difference for L2R and R2L calibration - Note that a car traveling at 50 kmh will pass the complete video frame in less than a second on the closer lane (L2R):
`# Calibration Settings
# --------------------
CALIBRATE_ON = False      # Create a calibration image file with calibration hash markers 10 px per mark
CAL_OBJ_PX_L2R = 115      # L2R Moving Objects, Length of a calibration object in pixels
CAL_OBJ_MM_L2R = 4789    # L2R Moving Objects, Length of the calibration object in millimetres
# L2R: 100 km/h => ca. 27,7 m/s => ca. 677 PX/s
# L2R:  50 km/h => ca. 27,7 m/s => ca. 338 PX/s
CAL_OBJ_PX_R2L = 65      # R2L Moving Objects, Length of a calibration object in pixels
CAL_OBJ_MM_R2L = 4789    # R2L Moving Objects, Length of the calibration object in millimetres
# R2L: 100 km/h => ca. 27,7 m/s => ca. 383 PX/s
# R2L:  50 km/h => ca. 27,7 m/s => ca. 192 PX/s
# Note if tested speed is too low increase appropriate cal_obj_mm  value and redo speed test for desired direction.
# IMPORTANT - If plugins Enabled Edit Settings in specified plugin file located in plugins folder.`

`# Motion Event Settings
# ---------------------
MO_SPEED_MPH_ON = False     # Set Speed Units   kph=False  mph=True
MO_TRACK_EVENT_COUNT = 4    # Default= 6 Number of Consecutive Motion Events to trigger speed photo. Adjust to suit.
                            # Suggest single core cpu=4-7 quad core=8-15 but adjust to smooth erratic readings due to contour jumps
MO_MIN_AREA_PX = 200        # Default= 200 Exclude all contours less than or equal to this sq-px Area
MO_LOG_OUT_RANGE_ON = True  # Default= True Show Out of Range Events per x_diff settings below False= Off
MO_MAX_X_DIFF_PX = 50       # Default= 20 Exclude if max px away >= last motion event x position
MO_MIN_X_DIFF_PX = 3        # Default= 1 Exclude if min px away <= last event x position
MO_X_LR_SIDE_BUFF_PX = 10   # Default= 10 Divides motion Rect x for L&R Buffer Space to Ensure contours are in
MO_TRACK_TIMEOUT_SEC = 0.7  # Default= 0.5 Optional seconds to wait after track End (Avoids dual tracking)
MO_EVENT_TIMEOUT_SEC = 0.1  # Default= 0.3 seconds to wait for next motion event before starting new track
MO_MAX_SPEED_OVER = 10       # Exclude track if Speed less than or equal to value specified 0=All
                            # Can be useful to exclude pedestrians and/or bikes, Etc or track only fast objects`

A small explanation what i do and why @pageauc your comments on these settings and my explanations are encouraged:
`MO_TRACK_EVENT_COUNT = 4`
`MO_MAX_X_DIFF_PX = 50` fast vehicles move many pixels/s on a fisheye lense, to avoid "Out" events i have increased this to fix many missed vehicles
`MO_MIN_X_DIFF_PX = 3` something that moves only 1 pixel seems very slow (and unlikely)
`MO_TRACK_TIMEOUT_SEC = 0.7` Put the time difference of two cars travelling at relatively high speed here. in my case the faster cars are reaching 100 kmh, with a distance of some 30 m between the cars the second car will reach the video frame in less than 1 second after the first car - i still track slow & long vehicles by accident though ... for example a bus ("coach"?) of 12m length traveling at 33 km/h (9m/s) will take roughly 1.5 seconds to pass the camera...
`MO_EVENT_TIMEOUT_SEC = 0.1`again, due to fisheye vehicles pass quickly and I think it's good to get the motion event "catcher" back in business as soon as possible to avoid missing any vehicles.

BR

konzty

closed 2023-11-06 by pageauc