pageauc/speed-camera #144: config.py still mentions configcam.py in header

konzty, 2023-10-12:
Hi,

as far as I understood all settings from configcam.py have been moved to config.py;

config.py still has the following comment in the header:
# IMPORTANT: Camera settings are now stored in configcam.py
#            This was done to allow camera code to be more portable

Should be removed, I guess.

BR

K

pageauc, 2023-10-12:
Latest update 13.13 has updated config.py and corrects comment.  config.py has been organized better.  After menubox.sh UPGRADE, you will need to copy the config.py.new to config.py to see the new settings and reorganized layout.

speed-cam.py is also upgraded with functions for getting motion contours, and getting biggest contour.  Still some work on converting function vars to snake case (not critical for code execution) so lower priority.
 
Also, I just added a feature to allow saving AI positive and negative images.  Once I have a few thousand I hope to figure out how to create an AI model for vehicle recognition.  Not sure what to use at this time Keras, PyTorch, Tensorflow, Etc.  I am still a newby with AI tool chains.

I gave up on haar cascades due to lots of false positives and having problems finding a good pretrained vehicle recognition model, so looking to create one myself with my own speed-cam 360x240 images.  Also parked vehicles caused issues.

In the longer term, I would also like to create a speed-cam flask web admin interface with features similar to menubox.sh

Comments are appreciated.

Regards Claude ...

konzty, 2023-10-14:
Hello Claude,

thank you for the quick fix and giving an insight in your thoughts and plans with the project, I havenothing to add to your AI ideas though, sorry.

From little what I know about AI image recognition is you'd first have to set the premises and identify your use case or target functionality - what is it the AI should recognize? Then it all comes down to your training data - it should be as diverse as possible. I'm thinking about color hues (or eliminiate color by going greyscale?), exposure times, camera type (IR, white balance).

But most importantly: Dont' listen to me, some random guy on the internet with no idea about training AI.

;-)

Ciao,

konzty

closed 2023-10-19 by konzty