pageauc/speed-camera #148: Cannot run speed-cam.py - states "Could not import cv2 library"

ninjadev64, 2023-10-24:
Hi there,

I'm trying to run speed-cam.py with Python 3.9.2 on a Raspberry Pi 4. I've installed OpenCV with `sudo apt install python3-opencv` and it has successfully installed itself. I've also installed it with Pip (`pip install opencv-python`) but that seems to have done nothing.

Additionally, the `python-` versions of all of the packages installed in the install script should be removed as they have all been replaced with `python3-` versions. This is the output of my run of the install script:

```
aman@octopi:~/speed-camera $ curl -L https://raw.github.com/pageauc/rpi-speed-camera/master/speed-install.sh | bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  8618  100  8618    0     0  25272      0 --:--:-- --:--:-- --:--:-- 25272
-----------------------------------------------
Upgrade speed-camera speed-install.sh ver 13.17
-----------------------------------------------
Upgrade Download GitHub Files
Note: config.py will not be overwritten. Updated settings are in config.py.new
menubox.sh                                    100%[================================================================================================>]  25.81K  --.-KB/s    in 0.01s   
speed-cam.py                                  100%[================================================================================================>]  83.62K  --.-KB/s    in 0.04s   
sql_speed_gt.py                               100%[================================================================================================>]   5.72K  --.-KB/s    in 0.006s  
speed-cam.sh                                  100%[================================================================================================>]   1.22K  --.-KB/s    in 0s      
search-speed.py                               100%[================================================================================================>]  13.83K  --.-KB/s    in 0.02s   
search_config.py                              100%[================================================================================================>]   2.00K  --.-KB/s    in 0.004s  
Readme.md                                     100%[================================================================================================>]  14.83K  --.-KB/s    in 0.01s   
makehtml.py                                   100%[================================================================================================>]   8.87K  --.-KB/s    in 0.009s  
webserver.py                                  100%[================================================================================================>]  11.79K  --.-KB/s    in 0.008s  
webserver.sh                                  100%[================================================================================================>]   1.29K  --.-KB/s    in 0.001s  
alpr-speed.py                                 100%[================================================================================================>]   7.31K  --.-KB/s    in 0.004s  
sql-make-graph-count-totals.py                100%[================================================================================================>]  10.66K  --.-KB/s    in 0.01s   
sql-make-graph-speed-ave.py                   100%[================================================================================================>]  10.74K  --.-KB/s    in 0.007s  
strmcam.py                                    100%[================================================================================================>]   7.34K  --.-KB/s    in 0.004s  
strmusbipcam.py                               100%[================================================================================================>]   1.66K  --.-KB/s    in 0.001s  
strmpilegcam.py                               100%[================================================================================================>]   2.59K  --.-KB/s    in 0.004s  
strmpilibcam.py                               100%[================================================================================================>]   2.73K  --.-KB/s    in 0.001s  
media/webserver.txt                           100%[================================================================================================>]   1.06K  --.-KB/s    in 0.002s  
config.py.new                                 100%[================================================================================================>]  12.36K  --.-KB/s    in 0.01s   
INFO  : Upgrade Check/Install pi-timolo/plugins    Wait ...
INFO  : __init__.py plugin Found.  Skip Download ...
INFO  : picam240.py plugin Found.  Skip Download ...
INFO  : webcam240.py plugin Found.  Skip Download ...
INFO  : picam480.py plugin Found.  Skip Download ...
INFO  : webcam480.py plugin Found.  Skip Download ...
INFO  : rtsp352.py plugin Found.  Skip Download ...
INFO  : picam720.py plugin Found.  Skip Download ...
INFO  : webcam720.py plugin Found.  Skip Download ...
INFO  : picam1080.py plugin Found.  Skip Download ...
INFO  : secpicam480.py plugin Found.  Skip Download ...
INFO  : secwebcam480.py plugin Found.  Skip Download ...
INFO  : Upgrade Check/Install speed-camera/rclone-samples    Wait ...
rclone-security-copy.sh                       100%[================================================================================================>]   4.26K  --.-KB/s    in 0.005s  
rclone-security-sync.sh                       100%[================================================================================================>]   4.26K  --.-KB/s    in 0.007s  
rclone-security-sync-recent.sh                100%[================================================================================================>]   4.26K  --.-KB/s    in 0.004s  
rclone-cleanup.sh                             100%[================================================================================================>]   1.04K  --.-KB/s    in 0.001s  
rclone v1.64.2
- os/version: raspbian 11.8
- os/kernel: 6.1.21-v7l+ (armv7l)
- os/type: linux
- os/arch: arm (ARMv7 compatible)
- go/version: go1.21.3
- go/linking: static
- go/tags: none
Upgrade Make required Files Executable
Upgrade Installing speed-cam.py Dependencies
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package python-opencv
Reading package lists...
Building dependency tree...
Reading state information...
[... 127 lines cut ...]

This is the error message that I get:

```
aman@octopi:~/speed-camera $ python3 speed-cam.py
Loading Wait...
----------------------------------------------------------------------
speed-cam.py 13.17  written by Claude Pageau
Motion Track Largest Moving Object and Calculate Speed per Calibration.
----------------------------------------------------------------------
2023-10-24 19:19:19 ERROR    <module>   Could Not import cv2 library
2023-10-24 19:19:19 ERROR    <module>   python3 failed to import cv2
2023-10-24 19:19:19 ERROR    <module>   Try installing opencv for python3
2023-10-24 19:19:19 ERROR    <module>   For RPI See https://github.com/pageauc/opencv3-setup
2023-10-24 19:19:19 ERROR    <module>   speed-cam.py 13.17 Exiting Due to Error
```

I tried the GitHub script mentioned in that error message but I get this error when running make compile and make install:

```
[ 22%] Building CXX object 3rdparty/carotene/hal/carotene/CMakeFiles/carotene_objs.dir/src/absdiff.cpp.o
c++: error: unrecognized command-line option ‘--param=ipcp-unit-growth=100000’; did you mean ‘--param=ipa-cp-unit-growth=’?
make[2]: *** [3rdparty/carotene/hal/carotene/CMakeFiles/carotene_objs.dir/build.make:82: 3rdparty/carotene/hal/carotene/CMakeFiles/carotene_objs.dir/src/absdiff.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3609: 3rdparty/carotene/hal/carotene/CMakeFiles/carotene_objs.dir/all] Error 2
```

ninjadev64, 2023-10-24:
(@pageauc)

pageauc, 2023-10-24:
I have Run the install on several different RPI's including Libre and
various RPI OS versions under python2 (older distros) and python3 under
bullseye 32 and 64 (RPI4)
without issue. Other users have not reported any install or opencv issues.
From the install script it is obvious that cv2 is installed.  Can you
import cv2 from a python command shell.

    python3
    >>import cv2
    >>print(cv2.__version__)
    >>exit()

You should get the python version number.  If not there may be a corrupt
python lib .so file or path problem.
This can happen with a download glitch.
It may also be a corrupt SD.  You should try a fresh SD card and burn a new
RPI OS image (Bullseye)  I have
not tested with bookworm yet but would not expect any problems.
Thanks for your detailed information
Claude ...

Claude ....

pageauc, 2023-10-24:
BTW  It is a good practice to perform a sudo apt update followed by a sudo
apt upgrade before doing any software install
Claude ....

pageauc, 2023-10-24:
FYI
Just created a bookworm 32 SD using pi Imager with predone settings for
WIFI, SSH, HOSTNAME, Etc.  Note I default initial hostname to rpi-noname so
it is easier to login with SSH.
inserted SD in RPI4.  Booted and after RPI initial setup that takes a bit
of time, I SSH Iogged in and did update and upgrade then did the curl
speed-camera install.
Had a RPI camera already installed.  Changed config.py to turn off
calibration since this was just for testing.  speed-cam.py ran fine with
default libcam setting.
No problems.

If I get time I will do bookworm 64 test
I suspect your SD has a problem.
Claude ....

ninjadev64, 2023-10-25:
> BTW  It is a good practice to perform a sudo apt update followed by a sudo
> 
> apt upgrade before doing any software install
> 
> Claude ....

Yes, I did do this (it took nearly 1 hour 😅)

ninjadev64, 2023-10-25:
Thanks for your quick response! I suspect my issue is to do with the "unrecognised command line parameter" error from make. I'll try importing it in a python shell later today.

ninjadev64, 2023-10-25:
I solved it!

I had it installed both with pip and with apt. There was something wrong with the pip version but it was still preferring it. 

`pip uninstall opencv-python` did the trick.

closed 2023-10-25 by ninjadev64