pageauc/speed-camera #149: No Window to view camera

victor7376, 2023-10-31:
Basically I installed on another raspberry to test the new changes and for some reason the camera view won’t open.  I’ve copied the settings over from the old version to the new one and got no camera view.

So I upgraded the old version to the latest on the working raspberry and still no camera window.

The settings I have are the following
GUI window on = true
GUI image win on = true

I have also tried thresh & crop to true but still no window of the camera even though the log says it’s opened up a window.

victor7376, 2023-10-31:
Btw… the program is still seeing the vehicles going past and recording the images and speeds.  I just don’t get a window with the live feed is what I mean.

johncblacker, 2023-11-02:
I'm running into the same problem...none of the GUI display setting seem to make a difference, no GUI windows yet images are being captured and I can view with webserver app...researching.

victor7376, 2023-11-02:
> OK, found the problem here: In module speed-cam.py, lines 1900-1956 must be indented otherwise they are outside the "while scanning..." loop and won't be executed to display the windows. Tested and verified.

can you post the file please?

victor7376, 2023-11-02:
Never mind I've fixed it, for anyone else between lines 1900 & 1954 add 4 spaces to each line then save & run the program

pageauc, 2023-11-03:
My mistake.  Github updated with fix
Claude

closed 2023-11-06 by pageauc