pageauc/speed-camera #160: How do I manually trigger the camera?

a113speederbot, 2024-02-24:
This is just a fantastic piece of work! Thank you so much to @pageauc for putting this awesome program together.

I know this is probably a very basic question but I can't find an answer to it: how do I trigger the camera manually, please?

Thanks!

pageauc, 2024-02-24:
There is a setting to align the camera. Not really manual like pressing a button but allows taking images at a selected frequency without motion tracking.  Image can be found in recent and is updated every specified seconds so use webserver refresh button to update image.  Default is False (Off) so you will need to edit config.py to enable.

        ALIGN_CAM_ON = True      # Default=False  True Saves alignment image to help with camera pointing
        ALIGN_DELAY_SEC = 3       # Default=3 seconds delay between each alignment image in recent folder

Claude ...

a113speederbot, 2024-02-24:
> There is a setting to align the camera. Not really manual like pressing a button but allows taking images at a selected frequency without motion tracking. Image can be found in recent and is updated [...]

Excellent, many thanks, Claude. One more thing (I hope), when changing settings in `config.py`, how do make them take effect? Stopping and starting `speed_cam.py` doesn't seem to do the trick. Should I be doing something else?

Thank you.

pageauc, 2024-02-24:
yes stop and restart speed-cam.py.  config.py settings updates will be used

a113speederbot, 2024-02-24:
Thank you, Claude - good to have that verified. Thank you for helping and being so friendly to somebody who's new at this. Probably I'll bother you some more but we can close this one. All the best to you - this is awesome!

closed 2024-02-24 by a113speederbot