pageauc/speed-camera #162: Capture being triggered by small objects and missing all vehicles.

a113speederbot, 2024-03-02:
Hello,

I changes the resolution of the camera to 720 using the plugin and resized the motion box in the `picam720.py` file. Since then, the camera seems only to be triggered by moving branches on trees, pedestrians, or if we pull one of our cars off the drive (very close to the camera). What have I missed, please?

Thank you.

SamDecrock, 2024-03-23:
I also have set my camera to 720p, here are my tweaks:

```
MO_TRACK_EVENT_COUNT = 7
MO_MIN_AREA_PX = 9000 
MO_MAX_X_DIFF_PX = 250
MO_MIN_X_DIFF_PX = 5  
IM_BIGGER = 1  
```

So yes, definitely increase `MO_MIN_AREA_PX` as you now have more pixels for the same area.

Also make sure your cropping area is bigger.

a113speederbot, 2024-09-14:
Thanks @SamDecrock, sorry - life got in the way somewhat so this reply is way overdue!

Just started trying to fix this issue again.

I have made changes to `picam720.py` like this:
```
MO_TRACK_EVENT_COUNT = 7
MO_MIN_AREA_PX = 9000
MO_LOG_OUT_RANGE_ON
MO_MAX_X_DIFF_PX = 250
MO_MIN_X_DIFF_PX = 5
MO_X_LR_SIDE_BUFF_PX = 10
MO_TRACK_TIMEOUT_SEC = 0.5
MO_EVENT_TIMEOUT_SEC = 0.3
MO_MAX_SPEED_OVER = 0
```
I have a module 3 camera and so I have changed the frame rate to 50fps in `strmpilibcam.py`

This is what the verbose log is saying:
```
Loading Wait...
----------------------------------------------------------------------
speed-cam.py 13.2  written by Claude Pageau
Motion Track Largest Moving Object and Calculate Speed per Calibration.
----------------------------------------------------------------------
[3:06:02.445034874] [6906]  INFO Camera camera_manager.cpp:316 libcamera v0.3.1+50-69a894c4
[3:06:02.453304751] [6914]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 29-08-2024 (16:33:32)
[3:06:02.463879439] [6914]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media2 and ISP device /dev/media1 using PiSP variant BCM2712_C0
[3:06:02.466513899] [6906]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
picamera2.picamera2 INFO: Camera configuration has been adjusted!
2024-09-14 18:54:13 INFO     configure_ Camera configuration has been adjusted!
[3:06:02.467407897] [6906]  INFO Camera camera.cpp:1191 configuring streams: (0) 1280x720-XRGB8888 (1) 4608x2592-BGGR_PISP_COMP1
[3:06:02.467567064] [6914]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected CFE format: 4608x2592-PC1B
picamera2.picamera2 INFO: Configuration successful!
2024-09-14 18:54:13 INFO     configure_ Configuration successful!
picamera2.picamera2 INFO: Camera started
2024-09-14 18:54:13 INFO     start_     Camera started
Logging Messages Disabled per LOG_VERBOSE_ON=False
----------------------------------------------------------------------
----------------------------------------------------------------------
----------------------------------------------------------------------
----------------------------------------------------------------------
----------------------------------------------------------------------```

Thanks :)

a113speederbot, 2024-09-15:
Ah, I just noticed this! `Logging Messages Disabled per LOG_VERBOSE_ON=False`