pageauc/speed-camera #163: Warnings in startup log: "Unsupported V4L2 pixel format"

a113speederbot, 2024-03-13:
Hi,

If I have `LOG_VERBOSE_ON = True` I get these warnings on startup. Are these something I need to address?

`[26:07:44.734741389] [4849]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16 
[26:07:44.734842351] [4849]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[26:07:44.734880814] [4849]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[26:07:44.734927166] [4849]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M`

Thanks!

SamDecrock, 2024-03-23:
In most (programming code related) cases, warnings can be ignored. Does it affect the application?

a113speederbot, 2024-09-15:
Thank you, @SamDecrock!

closed 2024-09-15 by a113speederbot