pageauc/speed-camera #164: Vehicles at higher speeds being missed.

a113speederbot, 2024-03-17:
Hi there,

I'm confident that everything is configured correctly, and largely it appears that speeds are being logged accurately (based on slow moving vehicles and pedestrians). The only problem I have is that it seems to be missing vehicles at higher speeds. My camera is fixed to the window itself, the lens isn't couching the window but is only a couple off mm away. The camera is about 9.5 m (31 ft) away from R2L and 12 m (39 ft)away from L2R. Is there anything glaringly wrong with this as regards physical positioning that you can see?

Thanks again for an awesome project.

SamDecrock, 2024-03-23:
You can increase `MO_MAX_X_DIFF_PX` so it's above the numbers you are seeing here 
```
Max D=30>=20px
Max D=67>=20px
Max D=30>=20px
Max D=30>=20px
```

brewtide, 2024-03-23:
Another aspect to this is what camera you are using to feed it the data. My homebrew situation is using RTSP to stream data from firmware modified wyze cams. These have a less than stellar frame rate which shows itself when dealing with higher speeds. The jump between version 1 of the cam and version 2 was large enough to make it effectively work. (I think wyze cam v1 is like 15 or 18FPS) and version 2 is nearing 25 FPS.

If you're running something like a picam, likely the framerate is not the problem.

a113speederbot, 2024-03-25:
> You can increase `MO_MAX_X_DIFF_PX` so it's above the numbers you are seeing here
> 
> ```
> Max D=30>=20px
> Max D=67>=20px
> Max D=30>=20px
> Max D=30>=20px
> ```
> 
> If cars move really fast, the difference in position between 'frames' is bigger. This variable is about that.

@SamDecrock, many thanks for your help here. Adjusting that setting certainly improves things, giving me many fewer 'Outs'. I'd be interested in what you'd suggest as setting adjustments; 10 px at a time, a few px at a time? Any ideas? I am also noticing lots of timeouts.

Thanks for explaining this to me.

closed 2025-03-02 by pageauc