pageauc/speed-camera #165: search-speed / Not compatible with the major rewrite?

brewtide, 2024-03-18:
Hello! 
Love your software. I recently went to use the search function to see how well / if it would work for my use case, and unfortunately it didn't seem to work. (I've attached the end result at bottom),

I did see that many configuration variables had changed (since the 13* version...) and while not a programmer, managed to change how it doesn't work, but yet to get the search function to work!

I realize it was not fully completed / beta, but I was wondering if you had any suggestions? I may try to setup an instance of the older version of the software, and configure my image-size / position units into the old config, but have the images pull from the 13.* image creation location to see if that will work.

Again, thanks for your software -- I figured I'd reach out and ask if you had any suggestions since you were the creator!
(And honestly, I'm not even sure if it 'worked' in the earlier speed-cam versions, but it seemingly did to some extent?)

search-speed.py ver 5.60 Loading  Please Wait .....
Traceback (most recent call last):
  File "/home/bbrewer/speed-camera/./search-speed.py", line 76, in <module>
    sw = 100 * image_bigger  # default search rectangle Width
               ^^^^^^^^^^^^
NameError: name 'image_bigger' is not defined

######################################
#          Review Output             #
######################################
  Press Enter to Return to Main Menu

SamDecrock, 2024-03-23:
I've not tested the code, but from the looks of it I think you can fix it by replacing `image_bigger` with `IM_BIGGER`.

brewtide, 2024-03-23:
Hey, I appreciate it. I kind of rambled in my above original question, but like you, I had done those additional variable changes and ended up with the same result, which is where I was left stumped. (Not really a programmer). I didn't paste that output as I had done it a week before and couldn't fully remember the 'new error' I was getting. 

I'll likely poke at this again this evening to see if I can figure it out -- Or I may just try to use one of the older versions (with old variables) but feed it my 'dataset'.

I was stuck at that point as well, and couldn't quite figure out steps beyond!

pageauc, 2025-03-02:
This has been fixed in the latest version as well as quite a few other updates.  Note background tasks are now managed by supervisor per install instructions, Readme.md and wiki.  Also supports non pi user accounts.  

I will be looking at search speed when I get some time.
Regards Claude ...

closed 2025-03-02 by pageauc