pageauc/speed-camera #169: Capture video

SamDecrock, 2024-04-15:
Is it possible to capture video at the same time?

The reason is that I want to review the speed measurements afterwards.

The only solution I can come up for this, is to turn my pi camera into a streaming camera, exposing it as an rtsp camera so I can both use it in speed-camera as record it.

aafuni, 2024-04-17:
I would also be interested if this is possible, to save the frames during the detection.

UnderpantsGnome, 2024-04-17:
I _think_ you should be able to do this by populating the [userMotionCode](https://github.com/pageauc/speed-camera/blob/master/user_motion_code.py) method, I have not tried this, but I was looking for a way to push data to an external API, but as the comment in there states `Note all functions and variables will be imported.`, I wasn't able to access any of the values being written to the DB or CSV.