pageauc/speed-camera #173: speed_cam.py process starts, then stops

spinko101, 2024-05-28:
Hi, 

I'm trying to get speed-camera to work on a RPI with Bookworm on a RPI 5. I'm using an rtsp ip camera and all is setup in the config.py file. When using ./menubox.sh the speed-cam.py process starts for a couple of seconds and then stops. 

Any guidance greatly appreciated.

spinko101, 2024-05-28:
----------------------------------------------------------------------
speed-cam.py 13.2  written by Claude Pageau
Motion Track Largest Moving Object and Calculate Speed per Calibration.
----------------------------------------------------------------------
2024-05-28 08:29:21 INFO     strmcam    Imported Required Camera Stream Settings from config.py
2024-05-28 08:29:21 INFO     is_pi_legacy_cam Check for Legacy Pi Camera Module with command - vcgencmd get_camera
2024-05-28 08:29:21 WARNING  is_pi_legacy_cam Command 'vcgencmd get_camera' returned non-zero exit status 255.
2024-05-28 08:29:21 ERROR    create_cam_thread import Failed. from strmpilibcam import PiLibCamStream

pageauc, 2025-03-02:
I suggest you start with a fresh install  of speed camera.  You can rename your current speed-camera version per
stop speed cam and webserver

    cd ~
    mv speed-camera speed-camera-prev

Install latest version from github curl command.
Follow instructions at end of install script.

The new speed-camera runs under a new method of running supervisorctl.  menubox.sh also runs under speed-cam.sh and speed-web.sh
Claude ...

Claude

closed 2025-03-02 by pageauc