pageauc/speed-camera #175: OpenCV and Python Version

djvdberg, 2024-08-30:
Getting below when starting ./speed-cam.py Where do I Start?

Running 32bit Bullseye

[image]

a113speederbot, 2024-09-22:
Have you tried installing opencv3? https://github.com/pageauc/opencv3-setup

pageauc, 2025-03-02:
I suggest you update to Bookworm or as a minimum Bullseye.  Latest version of speed-camera can run on libcamera / picamera2 python libraries if you are using a pi camera module.  picamera python library is only supported in in Bulleye (compatibility mode) and earlier.
web cameras and IP/RTSP streams are supported as usual.

Please note the latest version of speed-camera now uses supervisorctl to manage background processes per install instructions, Readme.md and wiki

Claude ...

closed 2025-03-02 by pageauc