pageauc/speed-camera #177: Any idea on why the tracks stop? The config seems okay as I've verified the speed with a radar gun.

benjaminfphillips, 2024-09-27:
I am using a RPI5 8GB with the Official Camera Module 3 Wide, which is static on a little tripod stand and the picam480 config. The only changes made were to bring the alignment down from where it was aimed and calibrate it at about 150px for the average car and UK cars are somewhere between 4400-4700mm in length.

The red car that was captured below was also verified at exactly 25mph using a radar gun, so in terms of overall speed tracking it seems okay but certain things it seems to start tracking and then give up on.

Running headless and just watching the logs via ssh. Nothing else, other than the web server are running.

Here is a snippet of some of the logs of my issue, it seems that fairly often, part of the way through the 6 tracks it'll just stop or lose track of the car? Then after some time (presumably when something else comes along) it'll start tracking that.

```

2024-09-27 13:46:18 INFO     speed_camera New  - 0/6 xy(358,0) Start New Track
2024-09-27 13:46:18 INFO     speed_camera  Add - 1/6 xy(374,0) 15.73 mph D=16/30 C=3 70x30=2100 sqpx L2R
2024-09-27 13:46:19 INFO     speed_camera  Add - 2/6 xy(390,0) 15.59 mph D=16/30 C=2 70x26=1820 sqpx L2R
2024-09-27 13:46:40 INFO     speed_camera Reset- event_timer 20.97>0.30 sec Exceeded

2024-09-27 13:50:26 INFO     speed_camera  Add - 1/6 xy(74,29) 18.66 mph D=19/30 C=2 184x91=16744 sqpx R2L
2024-09-27 13:50:26 INFO     speed_camera  Add - 2/6 xy(56,31) 18.14 mph D=18/30 C=3 186x89=16554 sqpx R2L
2024-09-27 13:50:35 INFO     speed_camera Reset- event_timer 9.06>0.30 sec Exceeded

2024-09-27 13:51:21 INFO     speed_camera New  - 0/6 xy(63,107) Start New Track
2024-09-27 13:51:21 INFO     speed_camera  Out - 0/6 xy(166,28) Max D=103>=30px C=7 27x9=243 sqpx L2R
2024-09-27 13:51:21 INFO     speed_camera  Add - 1/6 xy(151,29) 7.36 mph D=15/30 C=8 31x10=310 sqpx L2R
2024-09-27 13:51:21 INFO     speed_camera  Out - 1/6 xy(76,40) Max D=75>=30px C=5 22x13=286 sqpx L2R
2024-09-27 13:51:21 INFO     speed_camera  Add - 2/6 xy(104,35) 8.27 mph D=28/30 C=3 27x10=270 sqpx L2R
2024-09-27 13:51:21 INFO     speed_camera  Add - 3/6 xy(83,38) 9.18 mph D=21/30 C=3 41x10=410 sqpx L2R

2024-09-27 13:52:36 INFO     speed_camera New  - 0/6 xy(149,72) Start New Track
2024-09-27 13:52:36 INFO     speed_camera  Add - 1/6 xy(132,75) 16.71 mph D=17/30 C=2 51x39=1989 sqpx R2L
2024-09-27 13:52:36 INFO     speed_camera  Add - 2/6 xy(107,78) 20.59 mph D=25/30 C=3 55x42=2310 sqpx R2L
2024-09-27 13:52:36 INFO     speed_camera  Add - 3/6 xy(94,82) 16.71 mph D=13/30 C=3 48x38=1824 sqpx R2L
2024-09-27 13:52:36 INFO     speed_camera  Add - 4/6 xy(80,85) 15.23 mph D=14/30 C=2 41x35=1435 sqpx R2L
2024-09-27 13:52:37 INFO     speed_camera  Add - 5/6 xy(61,88) 16.71 mph D=19/30 C=2 40x32=1280 sqpx R2L
2024-09-27 13:52:43 INFO     speed_camera Reset- event_timer 6.48>0.30 sec Exceeded

2024-09-27 14:07:38 INFO     speed_camera New  - 0/6 xy(80,45) Start New Track
2024-09-27 14:07:38 INFO     speed_camera  Add - 1/6 xy(52,51) 13.76 mph D=28/30 C=4 91x55=5005 sqpx R2L
```

^^

[image]

Above is one of the captured images of where it has tracked and saved something over the 25mph limit I'd set.

The config is the picam480, with only the alignment area moved down as it was originally targeting the houses opposite.

Average sized car is about 150px and car sizes in the UK range from about 4400-4700mm roughly.
[image]

Then you have ones like this where the car is very likely to exceed my target.

2024-09-27 13:59:54 INFO     speed_camera New  - 0/6 xy(188,6) Start New Track
2024-09-27 13:59:54 INFO     speed_camera  Add - 1/6 xy(213,8) 24.55 mph D=25/30 C=5 46x25=1150 sqpx L2R
2024-09-27 13:59:54 INFO     speed_camera  Add - 2/6 xy(229,5) 20.08 mph D=16/30 C=6 47x30=1410 sqpx L2R
2024-09-27 13:59:54 INFO     speed_camera  Add - 3/6 xy(257,0) 24.55 mph D=28/30 C=8 41x22=902 sqpx L2R
2024-09-27 13:59:54 INFO     speed_camera  Add - 4/6 xy(281,0) 24.09 mph D=24/30 C=8 38x18=684 sqpx L2R
2024-09-27 14:00:58 INFO     speed_camera Reset- event_timer 63.47>0.30 sec Exceeded

but it also happens on ones where it the car is much slower, as in this example.

```
2024-09-27 14:01:28 INFO     speed_camera New  - 0/6 xy(348,0) Start New Track
2024-09-27 14:01:28 INFO     speed_camera  Add - 1/6 xy(372,0) 23.54 mph D=24/30 C=4 33x14=462 sqpx L2R
2024-09-27 14:01:28 INFO     speed_camera  Add - 2/6 xy(375,0) 13.24 mph D=3/30 C=2 33x13=429 sqpx L2R
2024-09-27 14:01:28 INFO     speed_camera  Add - 3/6 xy(387,0) 11.81 mph D=12/30 C=2 33x14=462 sqpx L2R
2024-09-27 14:01:28 INFO     speed_camera  Add - 4/6 xy(400,0) 12.29 mph D=13/30 C=1 23x12=276 sqpx L2R
2024-09-27 14:01:28 INFO     speed_camera  Add - 5/6 xy(414,0) 12.78 mph D=14/30 C=1 23x8=184 sqpx L2R
2024-09-27 14:01:33 INFO     speed_camera Reset- event_timer 5.50>0.30 sec Exceeded
```

It is the type of road where there can be a queue from the bottom which might enter the red area, but at this time of the day that isn't the case.

It either seems to go through all 6 adds smoothly or it does 2/3/4 and then pause, like this:

```
2024-09-27 14:06:22 INFO     speed_camera New  - 0/6 xy(76,51) Start New Track
2024-09-27 14:06:22 INFO     speed_camera  Add - 1/6 xy(97,47) 20.61 mph D=21/30 C=3 15x14=210 sqpx L2R
2024-09-27 14:06:22 INFO     speed_camera  Add - 2/6 xy(118,44) 20.60 mph D=21/30 C=4 16x13=208 sqpx L2R
2024-09-27 14:06:22 INFO     speed_camera  Add - 3/6 xy(139,41) 20.59 mph D=21/30 C=5 15x13=195 sqpx L2R
2024-09-27 14:06:22 INFO     speed_camera  Add - 4/6 xy(159,38) 20.46 mph D=20/30 C=7 15x14=210 sqpx L2R
2024-09-27 14:06:22 INFO     speed_camera  Add - 5/6 xy(180,36) 20.59 mph D=21/30 C=5 12x12=144 sqpx L2R
2024-09-27 14:06:31 INFO     speed_camera Reset- event_timer 8.78>0.30 sec Exceeded

2024-09-27 14:10:39 INFO     speed_camera New  - 0/6 xy(352,10) Start New Track
2024-09-27 14:10:39 INFO     speed_camera  Add - 1/6 xy(372,0) 19.57 mph D=20/30 C=5 26x24=624 sqpx L2R
2024-09-27 14:10:39 INFO     speed_camera  Add - 2/6 xy(386,0) 16.77 mph D=14/30 C=7 30x20=600 sqpx L2R
2024-09-27 14:10:39 INFO     speed_camera  Add - 3/6 xy(405,0) 18.38 mph D=19/30 C=4 28x17=476 sqpx L2R
2024-09-27 14:10:39 INFO     speed_camera  Add - 4/6 xy(422,0) 17.56 mph D=17/30 C=4 27x14=378 sqpx L2R
2024-09-27 14:10:42 INFO     speed_camera Reset- event_timer 2.51>0.30 sec Exceeded
```

benjaminfphillips, 2024-09-27:
I've uploaded a video (too big for here sadly) [That is about a minute long on imguir](https://imgur.com/a/NXvQuBu) where you can see it's done it at the start and a couple more times throughout.

a113speederbot, 2024-09-29:
Sorry, I can't help but since you have more or less an identical setup to me, I'd be interested in seeing your `config.py` file if you don't mind. I am using the same hardware as you save for that my module 3 cam is not the wide one and I am using the 720 plugin. The problem I am having is blurry images plus I can never get the camera to capture my vehicle so that I can verify calibration.

Would appreciate your help if you don't mind and sorry for the OT post!

benjaminfphillips, 2024-09-29:
If you want to maybe message me on discord? I've made a channel https://discord.gg/8mfZGAaR - that way, hopefully the main GH issue stays true to it's intent.

I notice I found it difficult to capture my car, but that was when I was INTENTIONALLY speeding through the target area. I've also now changed my calibration so that to the camera/code the line is straight, rather than a hill. I am hoping this means I can make the target area smaller.

I'm not sure if having a very long target area (red box) is a good or a bad thing or a bad thing, I'm starting to think it's necessary to capture FAST cars as they're gone before the app can even get the number of suggested points in many cases.

Did you try dropping down to the 480?

I have a feeling that opencv has too much processing to do, if your target are is too big. I/you could always test this by running the code on my pc 13900k, etc and a RTSP camera. There'a an iPhone app I downloaded and experimented with for £2.99 it was called IP Camera Pro, it's a black camera with a white wifi type symbol in the logo which is on a black background

https://apps.apple.com/gb/app/ip-camera-pro/id990605467

closed 2025-03-08 by pageauc