pageauc/speed-camera #179: Display Speed Only

kummerr, 2025-03-15:
Hi Claude,

I only want to display the speed only and not the file name on the capture image.  Is there a setting to change this?  If not, which file do I need to modify to get the desired result?

TIA

Joshude, 2025-05-13:
I wanted this too, so i just looked into the files. To change the overlay to speed only edit the `speed-cam.py`. Stop the service before editing it. After editing just start the process again. 

Line 1577 and the following contains this: 
```
                            if IM_SHOW_TEXT_ON:
                                image_text = "SPEED %.1f %s - %s" % (
                                    ave_speed,
                                    speed_units,
                                    filename,
                                )
```
Change to this to change to speed only: 
```
                            if IM_SHOW_TEXT_ON:
                                image_text = "SPEED %.1f %s" % (
                                    ave_speed,
                                    speed_units,
                                )
```

kummerr, 2025-05-13:
Thank you for sharing this! It would be nice if this could be an option in the settings so that it didn't get changed when an upgrade is applied.

kummerr, 2025-05-14:
Joshude, can you confirm the app still works after changing the speed-cam.py file?  Mine seems to have stop working even though the main menu shows the process is running.

Joshude, 2025-05-14:
The app works fine. You have to make sure that the format is right. So the trailing "," after "speed_units", all brackets and all the % on the right place.

kummerr, 2025-05-14:
I found my issue.  I neglected to modify the second line "image_text = "SPEED %.1f %s" % ("

Thanks again!